Fred78290/nct6687d #1: Not Working Upon Reboot

RedSoxFan04, 2020-11-26:
I installed the nct6687d kernel module last night and it was working beautifully. But when I booted up my PC this afternoon, the nct6687d kernel module was not functioning. Also, around the time that I shut down my PC, KDE Plasma started failing, with the background and taskbar disappearing.

Fred78290, 2020-11-26:
Hello, to load module at startup you must add it in file /etc/modules.

```
 # /etc/modules: kernel modules to load at boot time.
 #
 # This file contains the names of kernel modules that should be loaded
 # at boot time, one per line. Lines beginning with "#" are ignored.
 nct6687
```

Also I don't use KDE as WindowManager, and I can't help you about this side effect but probably some KDE modules registered some properties depending of lm-sensors lib or similar.

Fred78290, 2020-12-23:
Not really an issue

closed 2020-12-23 by Fred78290