Fred78290/nct6687d #3: Fan speed control WORKS with NCT6687

bmccann16, 2020-12-30:
I couldn't find your email address so I thought I'd report some positive testing results via an issue. No problems, just good status.

I'm running on Ubuntu 20.04 with kernel 5.4.0-56-generic on an MSI MPG B550i Gaming Edge motherboard which uses the NCT6687-R I/O controller. This machine is a file server and I have a script that monitors my disk drives and CPU temperatures and then generates a temperature value suitable for consumption for the Linux 'fancontrol' utility.

Fan control is working fine both with my script and during the calibration by the 'pwmconfig' utility.

Thank you for developing kernel module and please know that at least one person has used the fan control PWM successfully.

Fred78290, 2020-12-30:
Hello, many thanks for your report.
For fan control, I'm surprised and it's amazing. Probably I did a mistake during my test phase.

Please could you share your fan control script in this issue to update the README.

Thank's a lot.

bmccann16, 2020-12-30:
My script is probably too specific to publish in a README but I can refer you to the Linux utilities _pwmconfig_ and _fancontrol_. Pwmconfig is run once (by root) after loading the nct6687 Super I/O driver and it interacts with you, and your fans, to determine _which_ PWM output goes to which fan header on the motherboard. It also lets you select a Super I/O sensor, such as system temperature, to create a configuration file for the _fancontrol_ daemon. Here's the file that pwmconfig created for me:

INTERVAL=10
DEVPATH=hwmon5=devices/platform/nct6687.2592
DEVNAME=hwmon5=nct6687
FCTEMPS=hwmon5/pwm3=hwmon5/temp2_input
FCFANS= hwmon5/pwm3=hwmon5/fan3_input
MINTEMP=hwmon5/pwm3=32
MAXTEMP=hwmon5/pwm3=45
MINSTART=hwmon5/pwm3=20
MINSTOP=hwmon5/pwm3=20
MINPWM=hwmon5/pwm3=20

Note this is specific to my specific model of motherboard. My motherboard routes PWM #3 to the 'system' fan header.

This file is, by default, saved in /etc/fancontrol. The _fancontrol_ utility is typically run as a systemd daemon and it uses this info to monitor a sensor (identified by 'FCTEMPS') to automatically control fan speed. It senses the fan's speed using the sensor identified by FCFANS.

My personal fan control script synthesizes a temperature from the temperatures of six hard drives and the CPU temperature. It periodically writes this value to a file which fancontrol can read. You can point FCTEMPS at a specific file like this:

FCTEMPS=hwmon5/pwm3=/var/lib/fc-hddtemp/fc-hddtemp-average

Now, the _fancontrol_ utility will manage the fans based on that value. It can take some tweaking but the result (for me) is that the system fans controlled by the nct6687 ramp up when my CPU is hot _or_ when the drives get warm.