Fred78290/nct6687d #4: What's the difference between this driver and the upstream one?

gdamjan, 2021-01-25:
The Linux kernel got to support the nct6687d chip (in the nct6683 module) with 5.11 (now at -rc5)
https://github.com/torvalds/linux/commit/daf4fedde6177941b55ba3c3293a8585d5280b94

What is the difference between this and that driver?

Fred78290, 2021-01-25:
The main difference is the nct6683 is not the same chipset as nct6687, and command line sensor with the nct6683 module report wrong value for temp and voltage. I have tested it by recompiling the source nct6683.c from 5.10 on kernel 5.4 & 5.8.

Also temp/voltage labels are also wrong, the number of register aren't the same. The only thing working correctly is fan sensor.

About 5.11 I don't have tested yet, but the source code nct6683.c looks like the same.

gdamjan, 2021-01-25:
Thanks. I'm gonna compile this module too, and compare the results, and paste them here.

gdamjan, 2021-01-25:
these were taken in a 2 second period

nct6683.ko
```
nct6687-isa-0a20
Adapter: ISA adapter
VIN0:                992.00 mV (min =  +0.00 V, max =  +0.00 V)
VIN1:                  1.01 V  (min =  +0.00 V, max =  +0.00 V)
VIN2:                  1.09 V  (min =  +0.00 V, max =  +0.00 V)
VIN3:                672.00 mV (min =  +0.00 V, max =  +0.00 V)
VIN4:                448.00 mV (min =  +0.00 V, max =  +0.00 V)
VIN5:                832.00 mV (min =  +0.00 V, max =  +0.00 V)
VIN6:                720.00 mV (min =  +0.00 V, max =  +0.00 V)
VIN7:                  1.52 V  (min =  +0.00 V, max =  +0.00 V)
VCC:                   3.38 V  (min =  +0.00 V, max =  +0.00 V)
VTT:                   1.82 V  (min =  +0.00 V, max =  +0.00 V)
VSB:                   3.31 V  (min =  +0.00 V, max =  +0.00 V)
AVSB:                  3.31 V  (min =  +0.00 V, max =  +0.00 V)
VTT:                   1.82 V  (min =  +0.00 V, max =  +0.00 V)
VBAT:                  3.15 V  (min =  +0.00 V, max =  +0.00 V)
VREF:                  0.00 V  (min =  +0.00 V, max =  +0.00 V)
cpu fan:              575 RPM  (min =    0 RPM)
fan2:                   0 RPM  (min =    0 RPM)
system fan1 (rear):     0 RPM  (min =    0 RPM)
system fan2 (front):    0 RPM  (min =    0 RPM)
fan5:                   0 RPM  (min =    0 RPM)
fan6:                   0 RPM  (min =    0 RPM)
fan7:                   0 RPM  (min =    0 RPM)
fan8:                   0 RPM  (min =    0 RPM)
fan9:                   0 RPM  (min =    0 RPM)
fan10:                  0 RPM  (min =    0 RPM)
AMD TSI Addr 98h:     +33.0°C  (low  =  +0.0°C)
                               (high =  +0.0°C, hyst =  +0.0°C)
                               (crit =  +0.0°C)  sensor = AMD AMDSI
Diode 0 (curr):       +37.5°C  (low  =  +0.0°C)
                               (high =  +0.0°C, hyst =  +0.0°C)
                               (crit =  +0.0°C)  sensor = thermal diode
Thermistor 15:        +32.0°C  (low  =  +0.0°C)
                               (high =  +0.0°C, hyst =  +0.0°C)
                               (crit =  +0.0°C)  sensor = thermistor
Thermistor 1:         +41.0°C  (low  =  +0.0°C)
                               (high =  +0.0°C, hyst =  +0.0°C)
                               (crit =  +0.0°C)  sensor = thermistor
Thermistor 16:        +31.0°C  (low  =  +0.0°C)
                               (high =  +0.0°C, hyst =  +0.0°C)
                               (crit =  +0.0°C)  sensor = thermistor
Thermistor 0:         +35.0°C  (low  =  +0.0°C)
                               (high =  +0.0°C, hyst =  +0.0°C)
                               (crit =  +0.0°C)  sensor = thermistor
intrusion0:          ALARM
beep_enable:         disabled
```

nct6687.ko
```
nct6687-isa-0a20
Adapter: ISA adapter
+12V:                 12.10 V  (min = +12.07 V, max = +12.10 V)
+5V:                   5.06 V  (min =  +5.03 V, max =  +5.06 V)
+3.3V:                 3.33 V  (min =  +0.00 V, max =  +3.33 V)
CPU Soc:               1.10 V  (min =  +1.09 V, max =  +1.10 V)
CPU Vcore:           396.00 mV (min =  +0.40 V, max =  +0.74 V)
CPU 1P8:               1.83 V  (min =  +1.83 V, max =  +1.83 V)
CPU VDDP:              0.00 V  (min =  +0.00 V, max =  +0.00 V)
DRAM:                  1.36 V  (min =  +1.36 V, max =  +1.36 V)
Chipset:             836.00 mV (min =  +0.83 V, max =  +0.84 V)
cpu fan:              576 RPM  (min =  573 RPM, max =  809 RPM)
Pump Fan:               0 RPM  (min =    0 RPM, max =    0 RPM)
system fan1 (rear):     0 RPM  (min =    0 RPM, max =    0 RPM)
system fan2 (front):    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #3:          0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #4:          0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #5:          0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #6:          0 RPM  (min =    0 RPM, max =    0 RPM)
CPU:                  +33.0°C  (low  = +33.0°C, high = +39.0°C)
System:               +38.0°C  (low  = +37.0°C, high = +38.0°C)
VRM MOS:              +32.0°C  (low  = +32.0°C, high = +32.0°C)
PCH:                  +41.0°C  (low  = +41.0°C, high = +41.0°C)
CPU Socket:           +31.0°C  (low  = +31.0°C, high = +31.0°C)
PCIe x1:              +35.0°C  (low  = +35.0°C, high = +35.0°C)
M2_1:                  +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
```

module loading info:
```
[ 8902.471113] nct6687: Found NCT6687D or compatible chip at 0x004e:0x0a20
[ 8902.472384] nct6687 nct6687.2592: NCT6687D EC firmware version 1.0 build 05/07/20
```

Fred78290, 2021-01-25:
As you can see, voltage are not correct, values & labels
And in you report something goes wrong:
```
system fan1 (rear):     0 RPM  (min =    0 RPM, max =    0 RPM)
system fan2 (front):    0 RPM  (min =    0 RPM, max =    0 RPM)
```

It's not correct labels provided by the module. It must be:

```
System Fan #1:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #2:    0 RPM  (min =    0 RPM, max =    0 RPM)

```

gdamjan, 2021-01-25:
> It's not correct labels provided by the module. It must be:

I have some label overrides in `/etc/sensors.d/msi-mortar.conf` (should've removed the file before pasting).

gdamjan, 2021-01-28:
ps. which m2 slot is m2_1? because I do have a nvme in the m2_1 slot, the one connected to the cpu. it reports its own temperature, but here it says 0⁰.

also, 
do you have a plan of upstreaming the module?

graysky2, 2021-06-26:
@gdamjan - Thanks for the work on this.  I have the same question:

> do you have a plan of upstreaming the module?

berniyh, 2022-08-28:
> The main difference is the nct6683 is not the same chipset as nct6687, and command line sensor with the nct6683 module report wrong value for temp and voltage. I have tested it by recompiling the [...]

The upstream drivers usually don't apply scaling factors to voltages, especially for newer hardware.
It is possible to submit the configuration files upstream, but as long as nobody did that, the output will be incorrect.
If you apply the same multipliers (12 for in0, 5 for in1, 2 for in3 and 3 for in9), the voltages will be the same (obviously).
(Whether they are correct is another question, but it seems about right for my system.)

I've attached my configuration file as an example for those who are interested. (Remove the .txt at the end, only needed for uploading the file.)
Note though, that I changed the fan labels to my usage of the connectors and ignored those that I'm not using.

I'm not yet sure what to do with the additional voltage registers
in6-in8 and in12-in14
Also didn't set min/max/low/high/hyst/crit values yet, since I'm still in the process of verifying the data.

[nct6687D.txt](https://github.com/Fred78290/nct6687d/files/9439650/nct6687D.txt)
.

closed 2024-03-04 by Fred78290