Fred78290/nct6687d #5: Not compatible wit Kernel 5.11

DoXer67, 2021-02-18:
I hope you update the module for Linux 5.11
Thank you.

Fred78290, 2021-02-18:
Hi,

I'll update when I'll be able to install Linux 5.11 on Ubuntu TLS 20.04-2 without lost NVIDIA modules and running with 800x600 resolution :)

berniyh, 2021-02-27:
> I hope you update the module for Linux 5.11
> Thank you.

Works for me?
(On 5.11.0)

berniyh, 2021-02-27:
Maybe the problem was just that the kernel automatically loaded the nct6683 driver and thus the driver listed here fails (because the device is in use)?
In that case all you would have to do is to blacklist the nct6683 module.

Fred78290, 2021-02-27:
> Maybe the problem was just that the kernel automatically loaded the nct6683 driver and thus the driver listed here fails (because the device is in use)?
> In that case all you would have to do is to blacklist the nct6683 module.

Probably the true reason why it doesn't load, because modules using the same hardware are exclusive by using a lock.

TheMangoTrain, 2021-05-15:
Just upgrading to kernal 5.11 didn't make any change in sensor output --- my fans still not detected.

Running this github project, now everything works!

Many thanks!

pavilalopes, 2021-07-17:
Works for me on Ubuntu 21.04 with kernel 5.11.0

closed 2022-02-12 by Fred78290