Fred78290/nct6687d #6: T_SEN1 input missing

berniyh, 2021-02-27:
Hi,

thanks for the driver, it works very well on my MSI B550 Gaming Carbon board.
However, I noticed that one temperature is missing and that is the temperature for the T_SEN1 input, which can be used to monitor e.g. the case temperature or the water temperature in case of water cooling.
I have installed a thermister to the correct pin and in the BIOS I get a proper reading for the sensor (about 30°C).

I already tried increasing the number of temperatures being read in the driver to 8 (which is the maximum as far as I understood), but that gives me a 0°C reading, just like the M2 one.

Do you have an idea how to fix this or how to find out whether the thermister is connected to this controller?
I didn't see any other controller on the mainboard it could be connected to but it is of course possible that I missed something.

Fred78290, 2021-02-27:
Hello,

Do you have the ability to run a Windows 1O and test with Hardware monitor if the sensor is detected and display the good value?

berniyh, 2021-02-27:
No, unfortunately not. This PC is linux only.
But I've made a request for that already in a pc-oriented forum I'm participating in.
Don't know if anyone there has this board, though.

However, I just found this here, which could be helpful:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/361

FocalDE, 2021-06-01:
@Fred78290 @berniyh
Gents, 
attached a screenshot of HWInfo, which is detecting it just fine.
[image]

FocalDE, 2021-06-01:
@Fred78290 : no Luck with CPUID HW Monitor - shall i try something else?

Fred78290, 2021-06-06:
@FocalDE 
Thank's. It means that whitout public official documentation from Nuvoton, Reverse engineering is the rule.

berniyh, 2021-06-06:
The screenshot says "NUC126". For that I found docs here:
https://www.nuvoton.com/export/resource-files/DS_NUC126_Series_EN_Rev1.05.pdf

There is also info about a temperature sensor in section 8.4.2.

Don't quite understand though why they would hook up that sensor to another controller, especially one that seems to fulfill a different purpose normally.

Cersei123, 2022-11-14:
There is a workaround for using this sensor.

1. Download Fan Control
https://github.com/Rem0o/FanControl.Releases

2. Download and do the instructions for the HWinfo Plugin
https://github.com/Rem0o/FanControl.HWInfo

3. ????

4. Use the sensor as you like in Fan Control