Fred78290/nct6687d #8: Worked then no longer installs after update

farahats9, 2021-06-03:
I am on Kubuntu 21.04 with latest available kernel 5.11.0-18-generic and my motherboard is MSI mag b550m bazooka. I installed it and it worked for me, all fans were detected.
Then today it doesn't work and when I try to install it again it gave me error about `flex` not found and `bison` not found so I installed these.
Now the error its giving me is:
```
$ sudo make install     
make -C /lib/modules/5.11.0-18-generic/build M= modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-18-generic'
  SYNC    include/config/auto.conf.cmd
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/util.o
  HOSTLD  scripts/kconfig/conf
make[2]: *** No rule to make target 'arch/x86/tools/relocs_32.c', needed by 'arch/x86/tools/relocs_32.o'.  Stop.
make[1]: *** [arch/x86/Makefile:210: archscripts] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-18-generic'
make: *** [Makefile:4: all] Error 2
```

The system may have updated the kernel version or something after it worked but I don't know how to make it work again, any help is appreciated.

Fred78290, 2021-06-03:
Hello, You must install the kernel linux dev tools and headers associated version.

farahats9, 2021-06-03:
I did use `apt-get install build-essential linux-headers-`uname -r` still not working, do I need anything else?

Fred78290, 2021-06-03:
is the used kernel is coming from ubuntu or you install it from another repo.

Few week ago, with ubuntu 20.04, it was unable to build the module because linux dev env was broken. 

The problem is not in source code but in the build env.

farahats9, 2021-06-03:
No its the official kernel from ubuntu packages. it was just working well in 5.11.0-17 kernel and after update and reboot this happened.

Fred78290, 2021-06-03:
well it means that something is broken in kernel build essential release 18.

I think that you must post the issue to ubuntu.

Regards

benediktkr, 2021-06-04:
i was getting the same `Exec format error` on the `5.11.0-17-generic` kernel from ubuntu. After upgrading to `5.11.0-18-generic` i am however getting a different error

```
$ sudo make install
make -C /lib/modules/5.11.0-18-generic/build M=/home/ben/github/nct6687d modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-18-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-18-generic'
sudo cp nct6687.ko /lib/modules/5.11.0-18-generic/kernel/drivers/hwmon/
sudo depmod
sudo modprobe nct6687
modprobe: ERROR: could not insert 'nct6687': Invalid argument
make: *** [Makefile:9: install] Error 1
```

as expected, i get the same error with `insmod`:

```
$ make 
make -C /lib/modules/5.11.0-18-generic/build M=/home/ben/github/nct6687d modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-18-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-18-generic'

$ sudo insmod nct6687.ko
insmod: ERROR: could not insert module nct6687.ko: Invalid parameters

```

this is revealed by `dmesg`:

```
[Fri Jun  4 15:21:51 2021] nct6687: disagrees about version of symbol _dev_info
[Fri Jun  4 15:21:51 2021] nct6687: Unknown symbol _dev_info (err -22)
```

which to me suggests that the module was compiled against headers for a different version of the kernel (with a different symbol table). that would suggest that the ubuntu's kernel-headers package is broken right?

benediktkr, 2021-06-04:
dug around some more, and progressed back to 

```
16:00 bear ~/github/nct6687d <main*> $ sudo make install
make -C /lib/modules/5.11.0-18-generic/build M=/home/ben/github/nct6687d modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-18-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-18-generic'
sudo cp nct6687.ko /lib/modules/5.11.0-18-generic/kernel/drivers/hwmon/
sudo depmod
sudo modprobe nct6687
modprobe: ERROR: could not insert 'nct6687': Exec format error
make: *** [Makefile:9: install] Error 1
```

```
16:01 bear ~/github/nct6687d <main*> $ sudo insmod ./nct6687.ko
insmod: ERROR: could not insert module ./nct6687.ko: Invalid module format
```

with this `dmesg`: 

```
[Fri Jun  4 16:01:04 2021] module: x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc 00000000e132133f, val ffffffffc0c6e13d
```

Fred78290, 2021-06-06:
As already said, the kernel build env coming from ubuntu is broken as often.

I did a build using official kernel from kernel.org inside a multipass VM and every thing work fine, except that the controller doesn't exists in the VM.

```
ubuntu@hirsute:~/Projects/nct6687d$ make
make -C /lib/modules/5.11.0-051100-generic/build M=/home/ubuntu/Projects/nct6687d modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-051100-generic'
  CC [M]  /home/ubuntu/Projects/nct6687d/nct6687.o
  MODPOST /home/ubuntu/Projects/nct6687d/Module.symvers
  CC [M]  /home/ubuntu/Projects/nct6687d/nct6687.mod.o
  LD [M]  /home/ubuntu/Projects/nct6687d/nct6687.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-051100-generic'
ubuntu@hirsute:~/Projects/nct6687d$ make install
make -C /lib/modules/5.11.0-051100-generic/build M=/home/ubuntu/Projects/nct6687d modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-051100-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-051100-generic'
sudo cp nct6687.ko /lib/modules/5.11.0-051100-generic/kernel/drivers/hwmon/
sudo depmod
sudo modprobe nct6687
modprobe: ERROR: could not insert 'nct6687': No such device
make: *** [Makefile:9: install] Error 1
```
with kernel from 21.04, I got error
```
ubuntu@hirsute:~/Projects/nct6687d$ make install
make -C /lib/modules/5.11.0-18-generic/build M=/home/ubuntu/Projects/nct6687d modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-18-generic'
  CC [M]  /home/ubuntu/Projects/nct6687d/nct6687.o
  MODPOST /home/ubuntu/Projects/nct6687d/Module.symvers
  CC [M]  /home/ubuntu/Projects/nct6687d/nct6687.mod.o
  LD [M]  /home/ubuntu/Projects/nct6687d/nct6687.ko
  BTF [M] /home/ubuntu/Projects/nct6687d/nct6687.ko
Skipping BTF generation for /home/ubuntu/Projects/nct6687d/nct6687.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-18-generic'
sudo cp nct6687.ko /lib/modules/5.11.0-18-generic/kernel/drivers/hwmon/
cp: cannot create regular file '/lib/modules/5.11.0-18-generic/kernel/drivers/hwmon/': Not a directory
make: *** [Makefile:7: install] Error 1
```

You can install original kernel following this [instruction](https://www.tipsonunix.com/2021/02/how-to-install-linux-kernel-5-11-in-ubuntu-linux-mint/)

Regards

Fred78290, 2021-06-06:
So after installing a fresh ubuntu desktop 21.04 from ISO on another PC (without nct6687) and I build the module without problem.
The insmod failed due the missing controller.

I think that the problem is inside in your installation

```
fboltz@fboltz-hirsute:~/Projects/nct6687d$ make install
make -C /lib/modules/5.11.0-18-generic/build M=/home/fboltz/Projects/nct6687d modules
make[1] : on entre dans le répertoire « /usr/src/linux-headers-5.11.0-18-generic »
  CC [M]  /home/fboltz/Projects/nct6687d/nct6687.o
  MODPOST /home/fboltz/Projects/nct6687d/Module.symvers
  CC [M]  /home/fboltz/Projects/nct6687d/nct6687.mod.o
  LD [M]  /home/fboltz/Projects/nct6687d/nct6687.ko
  BTF [M] /home/fboltz/Projects/nct6687d/nct6687.ko
Skipping BTF generation for /home/fboltz/Projects/nct6687d/nct6687.ko due to unavailability of vmlinux
make[1] : on quitte le répertoire « /usr/src/linux-headers-5.11.0-18-generic »
sudo cp nct6687.ko /lib/modules/5.11.0-18-generic/kernel/drivers/hwmon/
sudo depmod
sudo modprobe nct6687
modprobe: ERROR: could not insert 'nct6687': No such device
make: *** [Makefile:9 : install] Erreur 1
```

closed 2022-02-12 by Fred78290