Fred78290/nct6687d #12: MAG Z590 TORPEDO

marco70-ms, 2021-08-07:
Hi all
I compiled the 5.13.8 kernel module nct6687.
Please fix ATX +5V

`nct6687-isa-0a20
Adapter: ISA adapter
ATX +12V:     11.95 V  (min = +11.95 V, max = +11.98 V)
ATX +5V:       6.77 V  (min =  +6.76 V, max =  +6.78 V)
ATX +3.3V:     3.36 V  (min =  +0.00 V, max =  +3.37 V)
CPU Soc:     666.00 mV (min =  +0.66 V, max =  +1.32 V)
CPU Vcore:   680.00 mV (min =  +0.68 V, max =  +0.68 V)
DRAM:          1.33 V  (min =  +1.33 V, max =  +2.67 V)
Chipset:       1.05 V  (min =  +1.05 V, max =  +1.05 V)
PUMP_FAN1:    663 RPM  (min =  659 RPM, max = 1398 RPM)
SYS_FAN1:     605 RPM  (min =  594 RPM, max =  621 RPM)
SYS_FAN4:     821 RPM  (min =  816 RPM, max =  842 RPM)
SYS_FAN6:     839 RPM  (min =  817 RPM, max =  843 RPM)
CPU temp:     +43.0°C  (low  = +42.0°C, high = +74.0°C)
Sysytem:      +38.0°C  (low  = +38.0°C, high = +40.0°C)
VRM MOS:      +39.0°C  (low  = +39.0°C, high = +44.0°C)
PCH CHIP:     +46.0°C  (low  = +46.0°C, high = +46.0°C)
CPU Socket:   +41.0°C  (low  = +40.0°C, high = +49.0°C)
PCIe x1:      +23.0°C  (low  = +23.0°C, high = +23.0°C)
M2_1:          +2.0°C  (low  =  +2.0°C, high =  +3.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +43.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +43.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +42.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +43.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +41.0°C  (high = +100.0°C, crit = +100.0°C)
Core 4:        +41.0°C  (high = +100.0°C, crit = +100.0°C)
Core 5:        +41.0°C  (high = +100.0°C, crit = +100.0°C)
`
Thanks

Lowrida, 2022-05-24:
Same board, same problem with Mint 20.3 and kernel 5.14.0-1036. 
I made a correction file for it:  /etc/sensors.d/Z590.conf

```
chip "nct6687-isa-0a20"
  compute in1 @/1.4, @*1.4
```

Now it reports 4,95 volt. But would be nice if it gets fixed!

Fred78290, 2022-05-25:
Hello, by reading label in sensors output it look like that the module used to report voltage & fan speed is not mine also the label "in1" used in the  /etc/sensors.d/Z590.conf is not a label used in my module.

Regards

Lowrida, 2022-05-26:
> Hello, by reading label in sensors output it look like that the module used to report voltage & fan speed is not mine also the label "in1" used in the /etc/sensors.d/Z590.conf is not a label used in my module.
> 
> Regards

Go to /sys/class/hwmon/hwmon(x)/ and check which of the *_input files gives the faulty 5 volt. On my system that's in1_input. You can check the label name in *_label