Fred78290/nct6687d #14: Exec Format Error on Fedora 35

Mossy93, 2021-11-15:
Hello,

I noticed my fan speeds and voltages were not working recently. Now when I attempt to do `sudo modprobe nct6687` I get `modprobe: ERROR: could not insert 'nct6687': Exec format error`.

I tried to do `make install` again, but no luck.

Fred78290, 2021-11-15:
Hello,

Try make clean install.

Regards

Mossy93, 2021-11-15:
> Hello,
> 
> Try make clean install.
> 
> Regards

Gave me
```modprobe: ERROR: could not insert 'nct6687': Exec format error
make: *** [Makefile:9: install] Error 1```

Fred78290, 2021-11-15:
It's your first install or the error occured after OS upgrade?

It's look like a problem with your   install of GCC.

Mossy93, 2021-11-15:
This happened after an OS upgrade.

Fred78290, 2021-11-16:
[image]

Hello, I have installed a fresh VM with Fedora35 on vmware, so except that the io controller nc6687d is not present i didn't have problem.
I think that you something broken on your distribution.

Not concerning the module itself, i close the issue.

Regards

closed 2021-11-16 by Fred78290

Mossy93, 2021-11-21:
I just did a `dnf` package update today and it updated some of the kernel packages. Now it is working again.