Fred78290/nct6687d #15: Kernel patch

Semmelrocks, 2021-12-09:
Hi,

I'd created a Kernel patch for this module. Tested successfully on Kernel version 5.10.76.

[nct6687d.patch.txt](https://github.com/Fred78290/nct6687d/files/7681354/nct6687d.patch.txt)

Fred78290, 2021-12-14:
Hi all, thank you for contribution.

Is the patch remove the detection inside nct6683.c?

Regards

Semmelrocks, 2021-12-14:
I didn't changed anything there. The patch will add nct6687 to the kernel makefile and Kconfig as new option directly under nct6683.

Jeoshua, 2021-12-14:
I have 6683 blacklisted, both 6683 and this 6687 patch compiled as modules, and 6687 just works.

Jeoshua, 2021-12-19:
And I made it through the upgrade to 5.16rc4 without issue. It seems this Just Works.

helmchenlord, 2023-02-16:
I've created a patch for kernel 6.1.12
[nct6687d-6.1.12.patch.txt](https://github.com/Fred78290/nct6687d/files/10750526/nct6687d-6.1.12.patch.txt)

closed 2024-03-04 by Fred78290