Fred78290/nct6687d #17: MSI MAG Z590 TORPEDO works!  (5V is off by 2V)

Lowrida, 2022-01-19:
Only issue i have is the report on 5V that is off by 2 volts!

`+12V:           12.10 V  (min = +12.07 V, max = +12.12 V)
+5V:             6.93 V  (min =  +6.92 V, max =  +6.95 V)
+3.3V:           3.35 V  (min =  +3.34 V, max =  +3.35 V)
CPU Soc:       772.00 mV (min =  +0.64 V, max =  +1.33 V)
CPU Vcore:     674.00 mV (min =  +0.67 V, max =  +0.68 V)
CPU 1P8:         0.00 V  (min =  +0.00 V, max =  +0.00 V)
CPU VDDP:        0.00 V  (min =  +0.00 V, max =  +0.00 V)
DRAM:            1.77 V  (min =  +1.28 V, max =  +2.66 V)
Chipset:         1.04 V  (min =  +1.04 V, max =  +1.04 V)
`

Fancontrol works like a charm on Linux Mint 20.3 with kernel 5.13

Fred78290, 2022-01-19:
Hello, could you verify the voltage in the bios?

Lowrida, 2022-01-20:
Strangest thing... i went into the bios and i got all sorts of voltages to view, but no 5V. So maybe that's why it is off?? I don't know. But when i got back into Mint i directly noticed that fancontrol showed no fans, conky showed no fans and no voltages! So i checked if the module was still enabled, but it is gone! It is still in /etc/modules but it does not get loaded... tried 4 reboots!

Lowrida, 2022-01-20:
I went into the source folder of the module and did a 'make install' and it got loaded again and all works.
I remember that this afternoon i had a Mint update with a new version of the kernel. Could that be the reason? 
Do i have to do this every time i get an update of my kernel?

Fred78290, 2022-01-20:
Hello, yes each time the kernel is updated you need to rebuild the module because the module is located in directory ls /lib/modules/<kernel version>

Lowrida, 2022-01-20:
Ah ok, good to know. Anything on the 5V showing 6,9V?

Fred78290, 2022-01-20:
> Ah ok, good to know. Anything on the 5V showing 6,9V?

It's hard for me to reply because i don't have this motherboard and secondly you don't see this voltage in your bios dashboard.

also i don't use mint as linux distribution so it's possible that you have a sensor conf file in /etc doing some transformation on result of sensors cmd line.

Until you can't check real values in bios, I don't have clear way to find the problem.

regards

Lowrida, 2022-01-20:
I saw that inxi gave the same value at 5V..
`Sensors:   System Temperatures: cpu: 42.0 C mobo: 40.0 C gpu: radeon temp: 27 C 
           Fan Speeds (RPM): cpu: 561 mobo: 395 
           Voltages: 12v: 12.10 5v: 6.93 3.3v: 3.34 vbat: N/A 
`

I don't have any configuration or correction files for sensors or a /etc/sensors.d folder! So it shows what it finds.

Fred78290, 2022-01-21:
Normal inxi use sensors to collect infos.

Well I have no idea. Except mesure the voltage on your PSU.

regards

Lowrida, 2022-01-22:
The ports are al around 5.05 volts with minor difference.. 

[image]

nickcmaynard, 2024-12-02:
Can corroborate with a 3rd sample here - looks like the Z590 Torpedo might be a bit "funky".  Mine lists 6.44V on the 5V sensor.

Lowrida, 2024-12-03:
I created a file `/etc/sensors.d/z950.conf` with this in it:

```
#
# Sensors config for Z590

chip "nct6687-isa-0a20"
  compute in1 @/1.36, @*1.36
```

And now it shows 5.10 volt.