Fred78290/nct6687d #23: Misreports value of System Fan as being 0 once fancontrol has been started

Krish-sysadmin, 2022-02-20:
Hi. Thanks for making this! I am having an issue with this however. When fancontrol has not been started the output of ``sensors`` is this.

```nct6687-isa-0a20
Adapter: ISA adapter
+12V:           12.02 V  (min = +11.95 V, max = +12.05 V)
+5V:             5.10 V  (min =  +5.08 V, max =  +5.11 V)
+3.3V:           3.35 V  (min =  +0.00 V, max =  +3.36 V)
CPU Soc:       788.00 mV (min =  +0.76 V, max =  +0.85 V)
CPU Vcore:     448.00 mV (min =  +0.35 V, max =  +1.42 V)
CPU 1P8:         1.82 V  (min =  +1.81 V, max =  +1.82 V)
CPU VDDP:        0.00 V  (min =  +0.00 V, max =  +0.00 V)
DRAM:            1.20 V  (min =  +1.20 V, max =  +1.20 V)
Chipset:       824.00 mV (min =  +0.73 V, max =  +0.83 V)
CPU Fan:        907 RPM  (min =  835 RPM, max = 1916 RPM)
Pump Fan:       408 RPM  (min =  397 RPM, max = 1355 RPM)
System Fan #1:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #2: 9917 RPM  (min =    0 RPM, max = 11214 RPM) ---->> NOTE THIS
System Fan #3:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #4:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #5:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #6:    0 RPM  (min =    0 RPM, max =    0 RPM)
CPU:            +36.0°C  (low  = +34.0°C, high = +65.0°C)
System:         +42.0°C  (low  = +41.0°C, high = +45.0°C)
VRM MOS:        +36.0°C  (low  = +36.0°C, high = +41.0°C)
PCH:            +39.0°C  (low  = +39.0°C, high = +43.0°C)
CPU Socket:     +34.0°C  (low  = +34.0°C, high = +40.0°C)
PCIe x1:        +36.0°C  (low  = +35.0°C, high = +41.0°C)
M2_1:            +0.0°C  (low  =  +0.0°C, high =  +0.0°C)

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +36.9°C  

nvme-pci-2b00
Adapter: PCI adapter
Composite:    +40.9°C  (low  =  -0.1°C, high = +86.8°C)
                       (crit = +89.8°C)
Sensor 1:     +40.9°C  (low  = -273.1°C, high = +65261.8°C)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +52.0°C  

amdgpu-pci-3000
Adapter: PCI adapter
vddgfx:      731.00 mV 
vddnb:       799.00 mV 
edge:         +34.0°C  
power1:      1000.00 uW 
```

but as soon as I start it the fan noise gets considerably better, however, the fan is still running, while ``sensors`` misreports it as being on 0 RPM (stopped)

```nct6687-isa-0a20
Adapter: ISA adapter
+12V:           12.02 V  (min = +11.95 V, max = +12.05 V)
+5V:             5.10 V  (min =  +5.08 V, max =  +5.11 V)
+3.3V:           3.35 V  (min =  +0.00 V, max =  +3.36 V)
CPU Soc:       794.00 mV (min =  +0.76 V, max =  +0.85 V)
CPU Vcore:     982.00 mV (min =  +0.35 V, max =  +1.42 V)
CPU 1P8:         1.82 V  (min =  +1.81 V, max =  +1.82 V)
CPU VDDP:        0.00 V  (min =  +0.00 V, max =  +0.00 V)
DRAM:            1.20 V  (min =  +1.20 V, max =  +1.20 V)
Chipset:       828.00 mV (min =  +0.73 V, max =  +0.83 V)
CPU Fan:        889 RPM  (min =  835 RPM, max = 1916 RPM)
Pump Fan:       399 RPM  (min =  396 RPM, max = 1355 RPM)
System Fan #1:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #2:    0 RPM  (min =    0 RPM, max = 11428 RPM)
System Fan #3:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #4:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #5:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #6:    0 RPM  (min =    0 RPM, max =    0 RPM)
CPU:            +35.0°C  (low  = +34.0°C, high = +65.0°C)
System:         +42.0°C  (low  = +41.0°C, high = +45.0°C)
VRM MOS:        +36.0°C  (low  = +36.0°C, high = +41.0°C)
PCH:            +40.0°C  (low  = +39.0°C, high = +43.0°C)
CPU Socket:     +34.0°C  (low  = +34.0°C, high = +40.0°C)
PCIe x1:        +35.0°C  (low  = +35.0°C, high = +41.0°C)
M2_1:            +0.0°C  (low  =  +0.0°C, high =  +0.0°C)

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +36.4°C  

nvme-pci-2b00
Adapter: PCI adapter
Composite:    +40.9°C  (low  =  -0.1°C, high = +86.8°C)
                       (crit = +89.8°C)
Sensor 1:     +40.9°C  (low  = -273.1°C, high = +65261.8°C)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +52.0°C  

amdgpu-pci-3000
Adapter: PCI adapter
vddgfx:      881.00 mV 
vddnb:       799.00 mV 
edge:         +34.0°C  
power1:        2.00 mW 
```

Fred78290, 2022-02-21:
Hello could you detail the usage of the fancontrol tool.

Some people used it without problem to control fans speed. see first issue.

first time i used it, all fans gone to 0 and the cpu shutdown by reaching the heat limit 🤣

Krish-sysadmin, 2022-02-24:
I just enable the fancontrol tool using systemd.

Krish-sysadmin, 2022-02-28:
@Fred78290 Bump

Fred78290, 2022-02-28:
Well because fancontrol can set values to stop the fan, it's possible that your fancontrol config is not correct.

https://askubuntu.com/questions/22108/how-to-control-fan-speed

So I don't consider this issue as bug.

Krish-sysadmin, 2022-02-28:
That's probably it. I tend to use my PC nowadays with headphones and it's been not cleaned inside for dust for a couple of days so that might be something to look into. So I've lost motivation to look into this further. Also, might you want to upstream this module?

closed 2022-03-03 by Krish-sysadmin