Fred78290/nct6687d #30: Confirm work with MAG Z690 TOMAHAWK WIFI DDR4 (MS-7D32) with kernel 5.16

bangbang93, 2022-03-09:
Conflict with kernel nct6683, needs to `rmmod nct6683`

ricciolino, 2023-03-08:
Does this kernel module works on this MB ?
I checked the actual installed chipsets on the MB and I noticed they are **nuvoTon 3961S**, not **nct6687d** ... so I am wondering if this is expected to work even if they are different chipsets.

ricciolino, 2023-03-08:
> Does this kernel module works on this MB ?
> I checked the actual installed chipsets on the MB and I noticed they are nuvoTon 3961S, not nct6687d ... so I am wondering if this is expected to work even if they are different chipsets.

I just tried and I can confirm it works (in my case without doing any `rmmod nct6683`) :)