Fred78290/nct6687d #31: MSI X570 Carbon Max Wifi

atmafox, 2022-03-10:
So good news!  The module works.

Bad news is that the external temperature probe doesn't seem to -- despite showing up in the BIOS.  Six temperatures show up in BIOS:

- CPU
- System
- VRM MOS
- PCH
- CPU Socket
- External temperature probe (used to monitor my cooling loop)

However, seven show up on this driver and only five give valid results:

- CPU
- System
- VRM MOS
- PCH
- CPU Socket

The other two are:

- PCIe x1
- M2_1

What info can I provide to assist?

atmafox, 2022-03-15:
I tried extending the temperature sensor registers by 1 (to 8) and unfortunately it just shows 0s, so I don't think this is an easy one I can fix myself.

Fred78290, 2022-03-15:
Hello
It's hard to find information about theses 2 sensors. I'm not sure that theses sensors are controlled by the nuvonto controller 6687.

Regards

atmafox, 2022-03-15:
That's what I'd arrived at.  Thanks for confirming!  Maybe someday I'll give it a Windows install and try to figure out where MSI Dragon Control Center pokes to get the TSEN_1 value.

Thank you!
-Atma

closed 2022-03-15 by atmafox