Fred78290/nct6687d #36: voltage sensors no work fine

pipes80-collab, 2022-04-24:
this si my problem with this module:
dram voltage not reading fine
nct6687-isa-0a20
Adapter: ISA adapter
+12V:           12.19 V  (min = +12.19 V, max = +12.22 V)
+5V:             5.03 V  (min =  +5.03 V, max =  +5.04 V)
+3.3V:           3.34 V  (min =  +3.34 V, max =  +3.34 V)
CPU Soc:         1.11 V  (min =  +1.11 V, max =  +1.11 V)
CPU Vcore:     674.00 mV (min =  +0.67 V, max =  +0.68 V)
CPU 1P8:         0.00 V  (min =  +0.00 V, max =  +0.00 V)
CPU VDDP:        0.00 V  (min =  +0.00 V, max =  +0.00 V)
DRAM:            2.24 V  (min =  +2.24 V, max =  +2.24 V)
Chipset:       994.00 mV (min =  +0.99 V, max =  +0.99 V)
CPU Fan:          0 RPM  (min =    0 RPM, max =    0 RPM)
Pump Fan:      4411 RPM  (min = 4411 RPM, max = 4687 RPM)
System Fan #1:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #2:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #3:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #4:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #5:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #6:    0 RPM  (min =    0 RPM, max =    0 RPM)
CPU:            +27.0°C  (low  = +27.0°C, high = +28.0°C)
System:         +25.0°C  (low  = +25.0°C, high = +26.0°C)
VRM MOS:        +30.0°C  (low  = +30.0°C, high = +30.0°C)
PCH:            +21.0°C  (low  = +21.0°C, high = +21.0°C)
CPU Socket:     +25.0°C  (low  = +25.0°C, high = +25.0°C)
PCIe x1:        +26.0°C  (low  = +26.0°C, high = +26.0°C)
M2_1:           +26.0°C  (low  = +26.0°C, high = +26.0°C)

Fred78290, 2022-04-25:
> this si my problem with this module:
>
> dram voltage not reading fine
>
> nct6687-isa-0a20
>
> Adapter: ISA adapter
>
> +12V:           12.19 V  (min = +12.19 V, max = +12.22 V)
>
> +5V:             5.03 V  (min [...]

Hello what is your mobo?

pipes80-collab, 2022-04-25:
is a MSI z490 unify

pipes80-collab, 2022-06-14:
up

pipes80-collab, 2022-06-18:
now voltage cpu soc is not precise

Fred78290, 2022-06-27:
Did you blacklisted nct6683 module ?

pipes80-collab, 2022-06-27:
No, it's needed?

pipes80-collab, 2022-06-27:
> Did you blacklisted nct6683 module ?

it's not loaded when give lsmod

Fred78290, 2022-06-28:
Yes because the nct6683 is concurrent of nct6687 and probably the nct6683 is installed inplace of mine.
Values coming from nct6683 are lazy.

pipes80-collab, 2022-06-28:
I don't know, your module is installed, and nct6683 is not loaded in modules kernel

Fred78290, 2022-06-28:
As is, the module sample values in the nct6687 register every 100ms nothing else.
So i don't know where come the erroneous values.

Did you fill something in /etc/sensors3.conf or /etc/sensors.conf or checkit ?

pipes80-collab, 2022-07-04:
> Così com'è, i valori di campionamento del modulo nel registro nct6687 ogni 100 ms nient'altro. Quindi non so da dove vengono i valori errati.
> 
> Hai compilato qualcosa in /etc/sensors3.conf o /etc/sensors.conf o checkit ?

no, i have not compiled nothing on sensor3.conf, i have check now and it's not nct6687d present

pipes80-collab, 2022-07-14:
yes i have same situation, i attached screenshot
[image]

Fred78290, 2022-07-15:
Hello, Have you the ability to run W10 and install Hardware monitor using LibreHardwareMonitor and check the concerned voltage.

Regards

opal06, 2022-07-15:
> Hello, Have you the ability to run W10 and install Hardware monitor using LibreHardwareMonitor and check the concerned voltage.
> 
> Regards

Yes, that should be possible, but it may take me a few days as I'm quite busy at the moment. But hopefully I will be able to get back to you soon with that.

pipes80-collab, 2022-07-15:
this is screenshot

[image]

Fred78290, 2022-07-16:
Well I have also no linux machine 10850K with an MSI B460M Bazooka having NCT6687D and under linux I have the error.

I can invest now to find the problem

opal06, 2022-07-16:
So, I got around to run win10 as well and here are some screenshots:
[image]

This is LibreHardwareInfo under load. VCore value is different from nct6687 driver, but still wrong, which can be seen when compared to hwinfo:

[image]

This matches what's set in the BIOS, so I think hwinfo actually reads correctly.

I hope this can help you. If you need something else, feel free to ask.

pipes80-collab, 2022-07-16:
Today i'll try in load

pipes80-collab, 2022-07-16:
i have make a chinebench r23 test for read voltage.
screen librehardwaremonitor attached
[image]

Fred78290, 2022-07-17:
Hello thank you for your effort, I find finally the problem. In fact between AMD motherboard and Intel motherboard the sensor are not connected with the same register.

So the default register declaration is for AMD mobo.

To allow custom voltage configuration I have created a new branch **manual-voltage** and update the README to explain how to.
So before merge the pull request, could you test it ?

Thank you.

opal06, 2022-07-17:
I will try to get around to it this evening or tomorrow. Thank you so much for your efforts!

opal06, 2022-07-17:
So, I have built the new branch and installed it. It works fine, now the key is getting the profile correct. I have made a first attempt for the MSI z490i Unify, which I will attach here. I needed to do some computation, so it might be missreporting.

@pipes80-collab If you could test it, that would be great. There is obviously no guaranty that MSI wired the same sensors up between the z490 unify and the z490i, but it's close enough that it is worth a shot. It would be especially great if you could go and set a few manual, fixed vcore voltages in BIOS to see if my calculations are correct across a few voltages. Everything up to 1.4V should be safe, especially for short tests.

Note: As Github limits the upload to certain file types, I added .txt as extension, which should be removed before use.
[Z490i-unify.conf.txt](https://github.com/Fred78290/nct6687d/files/9128482/Z490i-unify.conf.txt)
.

Fred78290, 2022-07-18:
@opal06 
Try my sensors conf in sensors.d, I think that it work correctly for you mobo.

opal06, 2022-07-18:
> @opal06 Try my sensors conf in sensors.d, I think that it work correctly for you mobo.

No, unfortunately that does not seem to work properly:

nct6687-isa-0a20
Adapter: ISA adapter
+12V:          145.73 V  (min = +145.73 V, max = +146.30 V)
+5V:            24.90 V  (min = +24.85 V, max = +25.00 V)
VCore:           3.38 V  (min =  +3.37 V, max =  +3.38 V)
DIMM:            1.34 V  (min =  +1.34 V, max =  +1.35 V)
CPU I/O:         0.00 V  (min =  +0.00 V, max =  +0.00 V)
CPU SA:          0.00 V  (min =  +0.00 V, max =  +0.00 V)
+3.3V:           1.25 V  (min =  +1.24 V, max =  +1.25 V)
CPU Fan:        769 RPM  (min =  715 RPM, max =  769 RPM)
Pump Fan:      1996 RPM  (min = 1990 RPM, max = 2000 RPM)
System Fan #1:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #2:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #3:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #4:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #5:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #6:    0 RPM  (min =    0 RPM, max =    0 RPM)
CPU:            +58.0°C  (low  = +40.0°C, high = +65.0°C)
System:         +54.0°C  (low  = +54.0°C, high = +54.0°C)
VRM MOS:        +45.0°C  (low  = +44.0°C, high = +45.0°C)
PCH:            +62.0°C  (low  = +61.0°C, high = +62.0°C)
CPU Socket:     +39.0°C  (low  = +38.0°C, high = +39.0°C)

We can see that both +12V and +5V are too high, which is fixed by removing the compute for them. Same goes for Vcore, which is way too high, while DRAM is correct.
So maybe MSI wired the chip differently from z490 to b460... This might be due to different CRM components. Unfortunately, I cannot find the VRM specs for the b460m Bazooka. The z490i Unify uses 90A smart power stages from Renesas.

Fred78290, 2022-07-18:
`
+12V: 145.73 V (min = +145.73 V, max = +146.30 V)
+5V: 24.90 V (min = +24.85 V, max = +25.00 V)
 `
Input register look like good just remove the compute multiplier x12 x5 but it means that multipliers applied twice (another sensors conf ? )

` VCore: 3.38 V (min = +3.37 V, max = +3.38 V)`

This input look like +3.3V
` +3.3V: 1.25 V (min = +1.24 V, max = +1.25 V)`

Look like vcore.

pipes80-collab, 2022-07-18:
> No, unfortunately that does not seem to work properly:
>
> nct6687-isa-0a20
> Adapter: ISA adapter
> +12V:          145.73 V  (min = +145.73 V, max = +146.30 V)
> +5V:            24.90 V  (min = +24.85 V, [...]

I'll give mu result tonight, After your configuration

pipes80-collab, 2022-07-18:
I just tested the sensors under load, a comparison with cpu-x, I attach image
[image]

Fred78290, 2022-07-19:
> a perfect ignorant question, could there be some error in the numerical arrangement of the sensors? comparing 3 data
> 
> [image]

I don't know because the display order is depending of application