Fred78290/nct6687d #49: kernel 6.0

Makhno77, 2022-10-03:
Does the module work with the new 6.0 kernel?

Fred78290, 2022-10-03:
Hello,  I don't know because I don't have kernel 6.x installed

tmiland, 2022-10-06:
Works like expected on 6.0 kernel.

```
nct6687-isa-0a20
Adapter: ISA adapter
+12V:           12.07 V  (min = +12.02 V, max = +12.07 V)
+5V:             5.01 V  (min =  +5.00 V, max =  +5.01 V)
+3.3V:           3.35 V  (min =  +3.35 V, max =  +3.35 V)
CPU Soc:       1000.00 mV (min =  +1.00 V, max =  +1.01 V)
CPU Vcore:     964.00 mV (min =  +0.37 V, max =  +1.44 V)
CPU 1P8:         1.82 V  (min =  +1.82 V, max =  +1.82 V)
CPU VDDP:        0.00 V  (min =  +0.00 V, max =  +0.00 V)
DRAM:            1.34 V  (min =  +1.34 V, max =  +1.34 V)
Chipset:       850.00 mV (min =  +0.85 V, max =  +0.86 V)
CPU SA:        794.00 mV (min =  +0.79 V, max =  +0.81 V)
Voltage #2:      1.52 V  (min =  +1.52 V, max =  +1.52 V)
AVCC3:           3.32 V  (min =  +3.32 V, max =  +3.32 V)
AVSB:            3.35 V  (min =  +3.34 V, max =  +3.35 V)
VBat:            1.82 V  (min =  +1.82 V, max =  +1.82 V)
CPU Fan:        884 RPM  (min =  883 RPM, max = 2065 RPM)
Pump Fan:         0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #1:    0 RPM  (min =    0 RPM, max = 1791 RPM)
System Fan #2:    0 RPM  (min =    0 RPM, max = 1788 RPM)
System Fan #3:    0 RPM  (min =    0 RPM, max = 1807 RPM)
System Fan #4:    0 RPM  (min =    0 RPM, max = 1260 RPM)
System Fan #5:    0 RPM  (min =    0 RPM, max = 1310 RPM)
System Fan #6:    0 RPM  (min =    0 RPM, max = 1739 RPM)
CPU:            +51.0°C  (low  = +42.0°C, high = +69.0°C)
System:         +36.0°C  (low  = +33.5°C, high = +36.0°C)
VRM MOS:        +47.5°C  (low  = +46.5°C, high = +50.0°C)
PCH:            +37.5°C  (low  = +36.0°C, high = +37.5°C)
CPU Socket:     +37.0°C  (low  = +36.0°C, high = +38.5°C)
PCIe x1:        +34.5°C  (low  = +34.0°C, high = +34.5°C)
M2_1:            +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
```

HeinzDo1961, 2022-10-07:
I get an error message:
Error! Bad return status for module build on kernel: 6.0.0-arch1-1 (x86_64)
Consult /var/lib/dkms/nct6687d/1/build/make.log for more information.
make: *** [Makefile:32: dkms/install] Error 10

Make.log:
DKMS make.log for nct6687d-1 for kernel 6.0.0-arch1-1 (x86_64)
Fr 7. Okt 17:12:43 CEST 2022
make -C /lib/modules/6.0.0-arch1-1/build M=/var/lib/dkms/nct6687d/1/build modules
make[1]: Warnung: Kein Jobserver verfügbar: -j1 wird gesetzt. Fügen Sie der Ursprungsregel
„+“ hinzu.
make[1]: Verzeichnis „/usr/lib/modules/6.0.0-arch1-1/build“ wird betreten
  CC [M]  /var/lib/dkms/nct6687d/1/build/nct6687.o
  MODPOST /var/lib/dkms/nct6687d/1/build/Module.symvers
  CC [M]  /var/lib/dkms/nct6687d/1/build/nct6687.mod.o
  LD [M]  /var/lib/dkms/nct6687d/1/build/nct6687.ko
  BTF [M] /var/lib/dkms/nct6687d/1/build/nct6687.ko
Failed to parse base BTF 'vmlinux': -22
make[3]: *** [scripts/Makefile.modfinal:60: /var/lib/dkms/nct6687d/1/build/nct6687.ko] Fehler 1
make[3]: *** Datei „/var/lib/dkms/nct6687d/1/build/nct6687.ko“ wird gelöscht
make[2]: *** [scripts/Makefile.modpost:134: __modpost] Fehler 2
make[1]: *** [Makefile:1766: modules] Fehler 2
make[1]: Verzeichnis „/usr/lib/modules/6.0.0-arch1-1/build“ wird verlassen
make: *** [Makefile:24: dkms/build] Fehler 2

inspectorgadget03, 2023-01-04:
[image]

closed 2024-03-04 by Fred78290