Fred78290/nct6687d #52: Upstreaming this to Linux kernel

lourdas, 2023-01-06:
Hi,

I've successfully built the module with my MSI MAG Tomahawk Z790 Wifi DDR4 motherboard and seems to work fine, although there are a few differences with the coretemp-isa-0000 values reported by lm-sensor (my running kernel is 6.0.16). What I'd like to ask is if you have tried to upstream this driver to the official Linux kernel project. The author of the nct6683 module is Guenter Roeck <linux@roeck-us.net> as per the hwmon documentation in Linux 6.0.

What do you think?

Fred78290, 2023-01-17:
Hi sorry for long delay to reply because I have also many private enterprise repositories flooding my mailbox 😂
So, I have never contacted Gunther. May be I must do it….

lourdas, 2023-03-18:
Hey @Fred78290, did you have a chance to contact anyone about upstreaming this?

Fred78290, 2023-04-13:
A chance maybe…
Some official linux distribution already include my work

lourdas, 2023-04-13:
> A chance maybe… Some official linux distribution already include my work
Which one?

ImmortAlexGM, 2023-07-11:
https://www.phoronix.com/news/Linux-6.5-HWMON

Newest HWMON patches to kernel 6.5 bring support for NCT6687D.

lourdas, 2023-07-11:
The upstream patch is the one from this repository or different code?

ImmortAlexGM, 2023-07-12:
Looks like it is false alert.
I dug into the kernel sources https://git.kernel.org/pub/scm/linux/kernel/git/groeck/linux-staging.git/tree/drivers/hwmon/nct6683.c and found that driver already has support for NCT6687D but with different ID.
Kernel driver sources: #define SIO_NCT6687_ID		0xd590
Fred78290's sources: #define SIO_NCT6687D_ID 0xd592
In-kernel support was added at 2020-12-02.

ImmortAlexGM, 2023-09-11:
It was not a false alert...
Kernel 6.5.2, MSI MAG B560M BAZOOKA, in-kernel driver:

nct6687-isa-0a20
Adapter: ISA adapter
VIN0:             1.02 V  (min =  +0.00 V, max =  +0.00 V)
VIN1:             1.34 V  (min =  +0.00 V, max =  +0.00 V)
VIN2:           624.00 mV (min =  +0.00 V, max =  +0.00 V)
VIN3:           624.00 mV (min =  +0.00 V, max =  +0.00 V)
VIN4:           672.00 mV (min =  +0.00 V, max =  +0.00 V)
VIN5:             1.06 V  (min =  +0.00 V, max =  +0.00 V)
VIN6:             1.02 V  (min =  +0.00 V, max =  +0.00 V)
VIN7:             1.54 V  (min =  +0.00 V, max =  +0.00 V)
VCC:              3.36 V  (min =  +0.00 V, max =  +0.00 V)
VSB:              3.36 V  (min =  +0.00 V, max =  +0.00 V)
AVSB:             3.36 V  (min =  +0.00 V, max =  +0.00 V)
VTT:              1.06 V  (min =  +0.00 V, max =  +0.00 V)
VBAT:             3.18 V  (min =  +0.00 V, max =  +0.00 V)
VREF:             1.82 V  (min =  +0.00 V, max =  +0.00 V)
fan1:            767 RPM  (min =    0 RPM)
fan2:           2752 RPM  (min =    0 RPM)
fan3:           1089 RPM  (min =    0 RPM)
fan4:           1090 RPM  (min =    0 RPM)
fan5:              0 RPM  (min =    0 RPM)
fan6:              0 RPM  (min =    0 RPM)
fan7:              0 RPM  (min =    0 RPM)
fan8:              0 RPM  (min =    0 RPM)
fan9:              0 RPM  (min =    0 RPM)
fan10:             0 RPM  (min =    0 RPM)
PECI 0.0:        +30.5°C  (low  =  +0.0°C)
                          (high =  +0.0°C, hyst =  +0.0°C)
                          (crit =  +0.0°C)  sensor = Intel PECI
Diode 0 (curr):  +35.0°C  (low  =  +0.0°C)
                          (high =  +0.0°C, hyst =  +0.0°C)
                          (crit =  +0.0°C)  sensor = thermal diode
Thermistor 15:   +31.5°C  (low  =  +0.0°C)
                          (high = +127.0°C, hyst = +27.0°C)
                          (crit =  +0.0°C)  sensor = thermistor
PCH CHIP:        +43.0°C  (low  =  +0.0°C)
                          (high =  +0.0°C, hyst =  +0.0°C)
                          (crit =  +0.0°C)
Thermistor 16:   +31.0°C  (low  =  +0.0°C)
                          (high =  +0.0°C, hyst =  +0.0°C)
                          (crit =  +0.0°C)  sensor = thermistor
Thermistor 0:    +23.0°C  (low  =  +0.0°C)
                          (high =  +0.0°C, hyst =  +0.0°C)
                          (crit =  +0.0°C)  sensor = thermistor
Thermistor 1:    +24.5°C  (low  =  +0.0°C)
                          (high =  +0.0°C, hyst =  +0.0°C)
                          (crit =  +0.0°C)  sensor = thermistor
intrusion0:     OK
beep_enable:    disabled

lourdas, 2023-09-29:
Ok, running with latest 6.5.5 kernel, I found that enabling nct6683D and nct6675F and compatibles like in the screenshot:

[image]

it detects the sensor chip:

    nct6683: Found NCT6687D or compatible chip at 0x4e:0xa20
    nct6683 nct6683.2592: NCT6687D EC firmware version 1.0 build 08/16/22

The output from `sensors`:

    nct6687-isa-0a20
    Adapter: ISA adapter
    VIN0:            992.00 mV (min =  +0.00 V, max =  +0.00 V)
    VIN1:            992.00 mV (min =  +0.00 V, max =  +0.00 V)
    VIN2:            608.00 mV (min =  +0.00 V, max =  +0.00 V)
    VIN3:              1.39 V  (min =  +0.00 V, max =  +0.00 V)
    VIN4:            672.00 mV (min =  +0.00 V, max =  +0.00 V)
    VIN5:              1.33 V  (min =  +0.00 V, max =  +0.00 V)
    VIN6:            880.00 mV (min =  +0.00 V, max =  +0.00 V)
    VIN7:              1.52 V  (min =  +0.00 V, max =  +0.00 V)
    VCC:               3.33 V  (min =  +0.00 V, max =  +0.00 V)
    CPU Fan:            0 RPM  (min =    0 RPM)
    fan2:               0 RPM  (min =    0 RPM)
    Back Fan:        1477 RPM  (min =    0 RPM)
    fan4:               0 RPM  (min =    0 RPM)
    fan5:               0 RPM  (min =    0 RPM)
    fan6:               0 RPM  (min =    0 RPM)
    Front Upper Fan: 1498 RPM  (min =    0 RPM)
    Front Lower Fan: 1220 RPM  (min =    0 RPM)
    fan9:               0 RPM  (min =    0 RPM)
    fan10:              0 RPM  (min =    0 RPM)
    PECI 0.0:         +41.5°C  (low  =  +0.0°C)
                            (high =  +0.0°C, hyst =  +0.0°C)
                            (crit =  +0.0°C)  sensor = Intel PECI
    Diode 0 (curr):   +36.5°C  (low  =  +0.0°C)
                            (high =  +0.0°C, hyst =  +0.0°C)
                            (crit =  +0.0°C)  sensor = thermal diode
    Thermistor 15:    +38.0°C  (low  =  +0.0°C)
                            (high = +127.0°C, hyst = +27.0°C)
                            (crit =  +0.0°C)  sensor = thermistor
    PCH CHIP:         +49.0°C  (low  =  +0.0°C)
                            (high =  +0.0°C, hyst =  +0.0°C)
                            (crit =  +0.0°C)
    Thermistor 16:    +36.5°C  (low  =  +0.0°C)
                            (high =  +0.0°C, hyst =  +0.0°C)
                            (crit =  +0.0°C)  sensor = thermistor
    intrusion0:      OK
    beep_enable:     disabled

ImmortAlexGM, 2023-09-30:
What’s interesting is that your fans are named, mine is not, it's just fan1-fan10, both connected and not.

lourdas, 2023-09-30:
> What’s interesting is that your fans are named, mine is not, it's just fan1-fan10, both connected and not.

Yes, you can change that. Inside /etc/sensors.d you create a file with content similar to this:

    chip "nct6687-isa-0a20"
        label fan1 "CPU Fan"
        label fan3 "Back Fan"
        label fan7 "Front Upper Fan"
        label fan8 "Front Lower Fan"

Just make sure which fanX corresponds to each actual fan in your case.

Atemu, 2023-10-01:
Problems I have with the upstream module that are fixed by this module:

1. On my machine, I need to force load the module
2. Sensor and fan names aren't very descriptive by default
3. You can't set fan speeds

I think it's worth pursuing getting these adjustments into the upstream kernel @Fred78290.