Fred78290/nct6687d #53: Compatible with MAG B650 TOMAHAWK WIFI

reinerh, 2023-01-07:
dmesg:
```
[ 5247.629366] nct6687: Found NCT6687D or compatible chip at 0x004e:0x0a20
[ 5247.634771] nct6687 nct6687.2592: NCT6687D EC firmware version 0.0 build 08/31/22
```

sensors:
```
k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +41.9°C  
Tccd1:        +36.0°C  
Tccd2:        +35.4°C  

nvme-pci-0100
Adapter: PCI adapter
Composite:    +28.9°C  (low  =  -5.2°C, high = +89.8°C)
                       (crit = +93.8°C)

nct6687-isa-0a20
Adapter: ISA adapter
+12V:           12.07 V  (min = +12.07 V, max = +12.10 V)
+5V:             5.08 V  (min =  +5.08 V, max =  +5.09 V)
+3.3V:           3.34 V  (min =  +0.00 V, max =  +3.34 V)
CPU Soc:         1.35 V  (min =  +1.35 V, max =  +1.35 V)
CPU Vcore:       1.27 V  (min =  +0.78 V, max =  +1.28 V)
CPU 1P8:         0.00 V  (min =  +0.00 V, max =  +0.00 V)
CPU VDDP:        0.00 V  (min =  +0.00 V, max =  +0.00 V)
DRAM:            1.34 V  (min =  +1.34 V, max =  +1.35 V)
Chipset:       276.00 mV (min =  +0.23 V, max =  +0.28 V)
CPU SA:        830.00 mV (min =  +0.83 V, max =  +0.83 V)
Voltage #2:      1.54 V  (min =  +1.54 V, max =  +1.54 V)
AVCC3:           3.33 V  (min =  +3.33 V, max =  +3.34 V)
AVSB:            3.34 V  (min =  +0.00 V, max =  +3.34 V)
VBat:            2.04 V  (min =  +0.00 V, max =  +2.04 V)
CPU Fan:        591 RPM  (min =  591 RPM, max =  705 RPM)
Pump Fan:         0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #1:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #2:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #3:  733 RPM  (min =  729 RPM, max =  734 RPM)
System Fan #4:  691 RPM  (min =  684 RPM, max =  691 RPM)
System Fan #5:  707 RPM  (min =  703 RPM, max =  713 RPM)
System Fan #6:    0 RPM  (min =    0 RPM, max =    0 RPM)
CPU:            +41.0°C  (low  = +41.0°C, high = +49.0°C)
System:         +35.5°C  (low  = +35.0°C, high = +35.5°C)
VRM MOS:        +31.5°C  (low  = +31.5°C, high = +32.5°C)
PCH:            +35.5°C  (low  = +35.0°C, high = +35.5°C)
CPU Socket:     +31.0°C  (low  = +31.0°C, high = +31.5°C)
PCIe x1:        +79.5°C  (low  = +78.5°C, high = +80.5°C)
M2_1:            +0.0°C  (low  =  +0.0°C, high =  +0.0°C)

amdgpu-pci-0f00
Adapter: PCI adapter
vddgfx:        1.47 V  
vddnb:         1.35 V  
edge:         +38.0°C  
PPT:          61.03 W  
```

Though it doesn't show the temperature of the M2_1 slot (which contains my SSD), and the board has two more M2 slots.

Fred78290, 2023-01-17:
Depending motherboards the sensor are not connected to the same register. Without having this motherboard I’m not able to to fix the m2 sensor.

Background-Basil, 2023-07-21:
Hi both

I have this motherboard and am delighted to see it is supported - thank you.  

I'm running Fedora 38 but cannot get any sensor output.  I've followed the instructions:

Clone this repository
~$ git clone https://github.com/Fred78290/nct6687d
~$ cd nct6687d
Build & install package
~$ make akmod

To make it loaded after system boots

Just add nct6687 into /etc/modules

sudo sh -c 'echo "nct6687" >> /etc/modules'
Arch-Linux with systemd
sudo sh -c 'echo "nct6687" >> /etc/modules-load.d/nct6687.conf'

(tried both above) to no avail.  

Please excuse my inexperience but can you tell me where I'm going wrong please?

Many thanks in advance.