Fred78290/nct6687d #67: MSI PRO Z690-A DDR4 (MS-7D25)

bwsd0, 2023-04-17:
Hello,

Thank you for implementing this module. My mainboard (MSI PRO Z690-A DDR4) was flashed with [Dasharo (coreboot+UEFI) v1.1.1](https://docs.dasharo.com/variants/msi_z690/releases/#v111-2023-02-23) which may affect values of readings reported below.

## Hardware Info

<pre>
Mainboard: PRO Z690-A DDR4(MS-7D25)
CPU Model: 12th Gen Intel(R) Core(TM) i5-12600K
</pre>

## Kernel

<pre>
$ uname -rms
Linux 6.2.11 x86_64
</pre>

## Sensor Readings

<pre>
sensors nct6687-isa-0a20 | xclip -sel c
</pre>
<details>
<pre>
nct6687-isa-0a20
Adapter: ISA adapter
+12V:           12.17 V  (min = +12.14 V, max = +12.17 V)
+5V:             5.06 V  (min =  +5.05 V, max =  +5.06 V)
+3.3V:           3.36 V  (min =  +0.00 V, max =  +3.36 V)
CPU Soc:       522.00 mV (min =  +0.35 V, max =  +0.71 V)
CPU Vcore:     608.00 mV (min =  +0.61 V, max =  +0.61 V)
CPU 1P8:         0.00 V  (min =  +0.00 V, max =  +0.00 V)
CPU VDDP:        0.00 V  (min =  +0.00 V, max =  +0.00 V)
DRAM:            1.09 V  (min =  +0.70 V, max =  +1.43 V)
Chipset:       920.00 mV (min =  +0.91 V, max =  +0.92 V)
CPU SA:        908.00 mV (min =  +0.91 V, max =  +0.91 V)
Voltage #2:      1.53 V  (min =  +1.52 V, max =  +1.53 V)
AVCC3:           0.00 V  (min =  +0.00 V, max =  +0.00 V)
AVSB:            3.36 V  (min =  +0.00 V, max =  +3.36 V)
VBat:            1.06 V  (min =  +0.00 V, max =  +1.06 V)
CPU Fan:       1260 RPM  (min = 1224 RPM, max = 1295 RPM)
Pump Fan:         0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #1:  898 RPM  (min =  731 RPM, max =  909 RPM)
System Fan #2:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #3:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #4:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #5:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #6:    0 RPM  (min =    0 RPM, max =    0 RPM)
CPU:             +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
System:         +29.0°C  (low  = +28.0°C, high = +31.0°C)
VRM MOS:        +21.5°C  (low  = +21.0°C, high = +24.0°C)
PCH:             +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
CPU Socket:     +21.5°C  (low  = +21.5°C, high = +24.0°C)
PCIe x1:         +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
M2_1:            +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
</pre>
</details>

<pre>
coretemp-isa-0000 | xclip -sel c
</pre>
<details>
<pre>
Adapter: ISA adapter
Package id 0:  +20.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:        +15.0°C  (high = +80.0°C, crit = +100.0°C)
Core 4:        +16.0°C  (high = +80.0°C, crit = +100.0°C)
Core 8:        +17.0°C  (high = +80.0°C, crit = +100.0°C)
Core 12:       +14.0°C  (high = +80.0°C, crit = +100.0°C)
Core 16:       +17.0°C  (high = +80.0°C, crit = +100.0°C)
Core 20:       +17.0°C  (high = +80.0°C, crit = +100.0°C)
Core 28:       +20.0°C  (high = +80.0°C, crit = +100.0°C)
Core 29:       +20.0°C  (high = +80.0°C, crit = +100.0°C)
Core 30:       +18.0°C  (high = +80.0°C, crit = +100.0°C)
Core 31:       +18.0°C  (high = +80.0°C, crit = +100.0°C)
</pre>
</details>

miczyg1, 2023-04-20:
Dasharo v1.1.1 does not yet configure the NCT6687D hardware monitor, so incorrect readings are expected for now.

t-gh-ctrl, 2024-06-06:
+1: the module works on my Z690-A DDR4 (wifi), with Dasharo 1.1.3, on Qubes OS 4.2.1 (kernel 6.8.8-1) and allows setting pwm values / control fans manually.

<details>

<summary>
If it helps someone, here's a sample `/etc/fancontrol` for my two chassis fans (id 4 and 5), as for some reason `pwmconfig` doesn't write anything despite successfully  identifying/testing the fans.
</summary>

```
INTERVAL=4

DEVPATH=hwmon0=devices/platform/nct6687.2592
DEVNAME=hwmon0=nct6687

# both fans use the PCH temp as reference
FCTEMPS=hwmon0/pwm4=hwmon0/temp4_input hwmon0/pwm5=hwmon0/temp4_input

FCFANS=hwmon0/pwm4=hwmon0/fan4_input hwmon0/pwm5=hwmon0/fan5_input

MINTEMP=hwmon0/pwm4=48 hwmon0/pwm5=46
MAXTEMP=hwmon0/pwm4=58 hwmon0/pwm5=58

# fans begin to spin at ~50% duty and stop at a bit less than 50%
# -> add 10% for starting
MINSTART=hwmon0/pwm4=155 hwmon0/pwm5=155
MINSTOP=hwmon0/pwm4=135 hwmon0/pwm5=135

# fan4 (top fan) can stop completely but fan5 should continue (blowing onto the PCH chipset)
MINPWM=hwmon0/pwm4=0 hwmon0/pwm5=135

AVERAGE=hwmon0/pwm4=4 hwmon0/pwm5=4
```
</details>

other: on @miczyg1 's advice I tried @xaizek 's [smart-fan-iv branch](https://github.com/xaizek/nct6687d/tree/smart-fan-iv): it kind of works - I can set values to the various `*_auto_point_*` files but I ran into a few issues (maybe that's because it's still a WIP - happy to help/discuss those in the upcoming PR).

xaizek, 2024-06-09:
@t-gh-ctrl, I'm curious to know what kind of trouble you've encountered.  I've enabled issues in my fork, so you can provide details there as I'm not yet openning a PR.  Those changes are WIP because:
 * they lack documentation and commits aren't tidy
 * you can't set/see SmartFan mode via `pwm*_enable`
 * new `pwm*_*` files are shown unconditionally (should be hidden if auto mode isn't SmartFan IV)
 * original configuration related to new functionality isn't restored on module unloading

Setting temperature/PWM levels of auto points should work fine provided that auto mode is set to be SmartFan IV by firmware, but they are sensitive to the order in which they are set because values are constrained by the rest of the values and you're setting them one by one.  Maybe if manual mode is active, constraints shouldn't be enforced to allow settings things in any order, I went with checks on every write instead of on activation of auto mode at least for now because this way you can know which value is incorrect right away.