Fred78290/nct6687d #72: Compatible with MAG B650 Carbon WiFi

TeamFahQ, 2023-07-08:
```
[    1.536604] nct6687: Found NCT6687D or compatible chip at 0x004e:0x0a20
[    1.539617] nct6687 nct6687.2592: NCT6687D EC firmware version 0.0 build 08/31/22
```

```
nvme-pci-0200
Adapter: PCI adapter
Composite:    +31.9 C  (low  =  -0.1 C, high = +117.8 C)
                       (crit = +149.8 C)

nct6687-isa-0a20
Adapter: ISA adapter
+12V:           12.02 V  (min = +12.00 V, max = +12.05 V)
+5V:             5.02 V  (min =  +5.01 V, max =  +5.03 V)
+3.3V:           3.36 V  (min =  +0.00 V, max =  +3.37 V)
CPU Soc:         1.02 V  (min =  +1.02 V, max =  +1.02 V)
CPU Vcore:       1.10 V  (min =  +0.46 V, max =  +1.10 V)
CPU 1P8:         0.00 V  (min =  +0.00 V, max =  +0.00 V)
CPU VDDP:        0.00 V  (min =  +0.00 V, max =  +0.00 V)
DRAM:            1.10 V  (min =  +1.10 V, max =  +1.10 V)
Chipset:       322.00 mV (min =  +0.18 V, max =  +0.34 V)
CPU SA:        716.00 mV (min =  +0.72 V, max =  +0.90 V)
Voltage #2:      1.52 V  (min =  +1.52 V, max =  +1.52 V)
AVCC3:           3.30 V  (min =  +3.30 V, max =  +3.31 V)
AVSB:            3.36 V  (min =  +0.00 V, max =  +3.36 V)
VBat:            2.05 V  (min =  +0.00 V, max =  +2.05 V)
CPU Fan:       1097 RPM  (min = 1061 RPM, max = 1109 RPM)
Pump Fan:      1655 RPM  (min = 1643 RPM, max = 1671 RPM)
System Fan #1:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #2: 1173 RPM  (min = 1120 RPM, max = 1223 RPM)
System Fan #3: 1182 RPM  (min = 1110 RPM, max = 1183 RPM)
System Fan #4: 1113 RPM  (min = 1039 RPM, max = 1119 RPM)
System Fan #5:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #6:    0 RPM  (min =    0 RPM, max =    0 RPM)
CPU:            +37.0 C  (low  = +34.0 C, high = +39.0 C)
System:         +41.5 C  (low  = +30.0 C, high = +41.5 C)
VRM MOS:        +36.5 C  (low  = +25.0 C, high = +36.5 C)
PCH:            +42.0 C  (low  = +32.0 C, high = +42.0 C)
CPU Socket:     +30.0 C  (low  = +24.0 C, high = +30.0 C)
PCIe x1:        +78.0 C  (low  = +76.5 C, high = +80.0 C)
M2_1:            +0.0 C  (low  =  +0.0 C, high =  +0.0 C)

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +37.6 C  
Tccd1:        +29.5 C  
Tccd2:        +29.8 C  

nvme-pci-0500
Adapter: PCI adapter
Composite:    +33.9 C  (low  =  -0.1 C, high = +77.8 C)
                       (crit = +80.8 C)
Sensor 1:     +33.9 C  (low  = -273.1 C, high = +65261.8 C)
Sensor 2:     +37.9 C  (low  = -273.1 C, high = +65261.8 C)
```

Only one missing is M2_1. Any ideas or suggestions as to how I can fix it (I understand you may not have the board). 

Second thing, I would like to add this to the upstream linux kernel with your permission.

Fred78290, 2023-07-08:
Hello, check in your bios at boot if the value is supplied.

> Le 8 juil. 2023 à 14:11, traines ***@***.***> a écrit :
> 
> Only one missing is M2_1. Any ideas or suggestions as to how I can fix it (I understand you may not have the board).

TeamFahQ, 2023-07-09:
It doesn't appear to have it in bios; however, I can check the temp of both of my nvme drive under settings>Manage Disks and Volumes. Attaching some screenies. But I'd have to say, if it's not in bios, then it's probably not going to work in linux...
[Bios snap](https://drive.google.com/file/d/163dCQVjdJO5iFv1ldj1ROcrtm5Yd-GRF/view?usp=drive_link)
[Windows snap1](https://drive.google.com/file/d/1nFPqvWb70C9oBiA5ZiJ26e1NmOmGmYfy/view?usp=drive_link)
[Windows snap2](https://drive.google.com/file/d/1wbN-6ANon8fyb5Cds3uhYjXBnu95Ox8v/view?usp=drive_link)

TeamFahQ, 2023-07-09:
Here is my commit for kernel. I currently used branch 6.3, am thinking of adding it to branch 5.15 lts and 5.19 since some distros have not updated to more current branches.

Fred78290, 2023-07-09:
> It doesn't appear to have it in bios; however, I can check the temp of both of my nvme drive under settings>Manage Disks and Volumes. Attaching some screenies. But I'd have to say, if it's not in [...]

If the BIOS doesn't show thé registre, it means that any sensor is connected.

Underground Windows you see the température because it is via SMART protocol underground SATA/PATA/PCI channel.

TeamFahQ, 2023-07-09:
> If the BIOS doesn't show thé registre, it means that any sensor is connected.
>
> Underground Windows you see the température because it is via SMART protocol underground SATA/PATA/PCI channel.

That's what I thought and I'm ok with that. Are you ok with me trying to get this merge into linux kernel?

imaami, 2023-09-09:
The upstream Linux kernel already has partial support for NCT6687D in the nct6683 driver. I'm not sure what you mean by "merge into the linux kernel", but if you mean the actual upstream Linux kernel then you're unlikely to succeed. The process for upstreaming a driver is involved and strict. And like I said there's already some support, so they'd probably want that one to be extended, not a competing separate driver.