Fred78290/nct6687d #73: Compatible with MSI MAG X670E TOMAHAWK WIFI

icedream, 2023-07-31:
Seems to be similar situation as https://github.com/Fred78290/nct6687d/issues/53. There are 4 M.2 slots but only for one of them the temperature is reported (currently unpopulated so unsure if it's working beyond displaying 0).

I can confirm at least CPU voltage, CPU temperatures working properly though. CPU fan values seem to be working fine as well, but the case fans (plugged in as system fans 1 and 2 for back and front) seem to report incorrect values near the minimum. Not sure if that's a sensor or a parsing issue.

`pwmconfig` plots for PWM-RPM correlations:

**CPU fan:**  
[image]

**System fan 1:**  
[image]

**System fan 2:**  
[image]

`dmesg` output:

```
[48297.780531] nct6687: Found NCT6687D or compatible chip at 0x004e:0x0a20
[48297.785831] nct6687 nct6687.2592: NCT6687D EC firmware version 0.0 build 09/27/22
```

`sensors` output:

```
nct6687-isa-0a20
Adapter: ISA adapter
+12V:           12.22 V  (min = +12.19 V, max = +12.26 V)
+5V:             5.06 V  (min =  +5.05 V, max =  +5.08 V)
+3.3V:           3.36 V  (min =  +3.35 V, max =  +3.36 V)
CPU Soc:         1.02 V  (min =  +1.01 V, max =  +1.03 V)
CPU Vcore:       1.30 V  (min =  +0.77 V, max =  +1.38 V)
CPU 1P8:         1.84 V  (min =  +1.84 V, max =  +1.84 V)
CPU VDDP:        0.00 V  (min =  +0.00 V, max =  +0.00 V)
DRAM:            1.10 V  (min =  +1.10 V, max =  +1.11 V)
Chipset:       580.00 mV (min =  +0.57 V, max =  +0.65 V)
CPU SA:        558.00 mV (min =  +0.54 V, max =  +0.62 V)
Voltage #2:      1.54 V  (min =  +1.53 V, max =  +1.54 V)
AVCC3:           3.35 V  (min =  +3.34 V, max =  +3.35 V)
AVSB:            3.36 V  (min =  +3.35 V, max =  +3.36 V)
VBat:            2.04 V  (min =  +2.04 V, max =  +2.04 V)
CPU Fan:       1146 RPM  (min =  761 RPM, max = 1976 RPM)
Pump Fan:         0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #1:  797 RPM  (min =  778 RPM, max =  834 RPM)
System Fan #2:  829 RPM  (min =  808 RPM, max =  834 RPM)
System Fan #3:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #4:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #5:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #6:    0 RPM  (min =    0 RPM, max =    0 RPM)
CPU:            +54.0°C  (low  = +40.0°C, high = +93.0°C)
System:         +52.0°C  (low  = +46.5°C, high = +53.5°C)
VRM MOS:        +42.5°C  (low  = +37.0°C, high = +51.5°C)
PCH:            +53.5°C  (low  = +48.5°C, high = +54.5°C)
CPU Socket:     +39.0°C  (low  = +34.0°C, high = +54.0°C)
PCIe x1:        +51.5°C  (low  = +46.0°C, high = +52.5°C)
M2_1:            +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
```

nm17, 2024-03-08:
Everything is the same here. I also have a MSI MAG X670E Tomahawk WIFI.

cebarks, 2024-09-14:
I have 3 m.2 drives installed and see individual temps for each one.

chowdri, 2024-10-29:
I'm using kernel 6.8.0-47, however I'm unable to install this module through any of the installation methods listed. apt is unable to find dh-dkms for manual installation. 
I'm using X670E Tomahawk WIFI motherboard as well. 
any help with the installation process would be helpful. 

when running make dkms/install I get the following:

make: *** [Makefile:54: dkms/install] Error 10

Thanks for any help with the matter