Fred78290/nct6687d #76: Loading with /etc/modules-load.d/nct6687.conf doesn't work

ImmortAlexGM, 2023-08-25:
Arch Linux user here.
I do `sudo sh -c 'echo "nct6687" >> /etc/modules-load.d/nct6687.conf'` as suggested, but at loading time I get this message from driver:
kernel: nct6687: loading out-of-tree module taints kernel.
kernel: nct6687: module verification failed: signature and/or required key missing - tainting kernel
kernel: nct6687: EC base I/O port unconfigured
systemd-modules-load[322]: Failed to insert module 'nct6687': No such device
kernel: nct6683: Found NCT6687D or compatible chip at 0x4e:0xa20
And module doesn't work until I do `sudo modprobe nct6687`
Once upon a time (a couple of boots) it worked, but no more.

What does this error means? Why it works with manual probing?

Fred78290, 2023-08-26:
I think that you need to rebuild the module after upgrading dev tools

ImmortAlexGM, 2023-08-26:
It is installed with DKMS, so rebuilded regularly. And it is Arch, so there's nothing to upgrade more than it is.

Fred78290, 2023-08-26:
Sometimes the build-essential package is not up to date. as the the message say: the version of the buildin kernel is not the same a thé build-essential package to build kernel modules.

often i have the same problem on ubuntu.

ImmortAlexGM, 2023-08-27:
If you talk about 'module tainting kernel', it's not definitely a reason. Those times when module was loaded at boot time, this message appeared too, and gurus at Arch message board tells me that it's just a warning.
But next message, 'EC base I/O port unconfigured', is the indicator of some floating problem. But I have no enough knowledge in kernel programming to understand what's happening at that time and what can I do (or how I can configure boot process) to resolve this situation.