Fred78290/nct6687d #78: Can't give fan control back to the firmware

Atemu, 2023-10-21:
After setting fan speed manually, there is no obvious way to give fan control back to the motherboard firmware; applying the firmware fan curves again. It'd be great if you could i.e. write `-1` into the pwm device to do that.

SergeyMy, 2023-10-24:
There is NO way to return, it seems like it should return when unloading the module, but alas, there is no way.

Fred78290, 2023-10-24:
Hi all, the main problem is to collect original settings, save it and restore it as needed.

In bios you can set different level for speed fan according temperature. Via the NCT6667 register you just say run the fan as this speed, so with fancontrol when you ask to save the state, it will be the instant state not the settings from bios.

I think that the temperature/fan speed according the temperature that you set in the bios is not handled by NCT itself but with another helper controller.

Having reading NCT 6683 technical book, I haven't see the feature fan speed/temperature control.

In the official kernel the feature to change fan speed is not available.

Help welcome.

SergeyMy, 2023-10-25:
I just started looking yesterday... in kernel 6.5.8 :) such drivers have a control mode register manually or automatically....
I'll continue searching :)

SergeyMy, 2023-10-25:
> Help welcome.

static void nct6687_restore_fan_control
{
.....
nct6687_write(data, NCT6687_REG_FAN_CTRL_MODE(0), 0);
} //end static void nct6687_restore_fan_control :):):)
FAN_CTRL_MODE 
1 - manual
0 -auto

Fred78290, 2023-10-25:
Create another template group command allowing switch mode auto/manual for pwm

`
SENSOR_TEMPLATE(pwmode, "pwmmode%d", S_IRUGO, show_pw, switch_pwm, 0);
`
Why not

SergeyMy, 2023-10-25:
better by analogy with // adhere to the standard ;)
**Documentation/hwmon/nct6775.rst**
```
sysfs attributes
----------------

pwm[1-7]
    - this file stores PWM duty cycle or DC value (fan speed) in range:

           0 (lowest speed) to 255 (full)

pwm[1-7]_enable
    - this file controls mode of fan/temperature control:

        * 0 Fan control disabled (fans set to maximum speed)
        * 1 Manual mode, write to pwm[0-5] any value 0-255
        * 2 "Thermal Cruise" mode
        * 3 "Fan Speed Cruise" mode
        * 4 "Smart Fan III" mode (NCT6775F only)
        * 5 "Smart Fan IV" mode

pwm[1-7]_mode
    - controls if output is PWM or DC level

        * 0 DC output
        * 1 PWM output

```

PS pwm[1-7]_mode **DC/PWM**  is also in nct6687, all that remains is to find it....
PSPS where did you find out about NCT6687_REG_FAN_CTRL_MODE 0xA00 can you share the source/datasheet? but on NCT6683?

SergeyMy, 2023-10-26:
@Fred78290 
ничего не понимаю :( судя по коду nct6687_write(data, NCT6687_REG_FAN_CTRL_MODE(0), 0); излишне т.к. регистр сохранялся и должен был восстанавливатся, но этого не происходило.
сей час же все работает как задуманно. :(
посмотрел исходный код в ядре 6.6-rc7 там полный ужас.
So I’m thinking what to take as a basis? yours or... https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/?h=v6.6-rc7  or both :)

SergeyMy, 2023-11-03:
@Fred78290 
do you have a **register map nct6687** because it takes a long time to search through reverse engineering :(

Fred78290, 2023-11-03:
I have the same and my 5700G die, waiting for RAM.
.

SergeyMy, 2023-11-04:
NCT6687_REG_PWM_SOURCE_TEMP_VALUE(x) (0xca0 + (x))
:)

SergeyMy, 2023-11-04:
> I have the same and my 5700G die, **waiting for RAM**. .

I took an unnamed 2 to 16 from a Chinese supplier, happy as an elephant
[image]

PR #87 (2024-01-28): Add `pwm[1-8]_enable`
* Fix restoring automatic fan mode on module unload.
* Document available parameters and how to use writeable sysfs files.
* Add `pwm[1-8]_enable` to switch between manual/firmware-setup mode.

Closes #78.
closed 2024-01-28 by Fred78290