Fred78290/nct6687d #82: reverse engineering for nct6687

SergeyMy, 2023-11-09:
@Fred78290 How do you feel about reverse engineering?
I didn’t find documentation for 6687 and I think you don’t have it either, but there are some nuances :)

SergeyMy, 2023-11-13:
NCT6683 and NCT6687
// 0 - normal 1 - bypass_mode
`#define NCT6687_REG_FAN_CTRL_MODE 0xA00`

```
#define NCT6687_REG_FAN_PWM_COMMAND 0xA01
#define NCT6687_FAN_CFG_REQ 0x80
#define NCT6687_FAN_CFG_DONE 0x40

```
```
nct6687_write(data, NCT6687_REG_FAN_PWM_COMMAND, NCT6687_FAN_CFG_REQ);
msleep(50);
nct6687_write(data, NCT6687_REG_PWM_WRITE(index), val);
nct6687_write(data, NCT6687_REG_FAN_PWM_COMMAND, NCT6687_FAN_CFG_DONE);

```
NCT6687 -Ok
NCT6683 -Fail

```

nct6687_write(data, NCT6687_REG_FAN_PWM_COMMAND, NCT6687_FAN_CFG_REQ);
msleep(50);
nct6687_write(data, NCT6687_REG_PWM_WRITE(index), val);
// nct6687_write(data, NCT6687_REG_FAN_PWM_COMMAND, NCT6687_FAN_CFG_DONE); //return to auto-mode
nct6687_write(data, NCT6687_REG_FAN_PWM_COMMAND, 0x00); //manual mode
```
NCT6683 -OK!

SergeyMy, 2023-11-13:
based on above...functions
save_fan_control
restore_fan_control
not only do not make sense (REG_FAN_CTRL_MODE
REG_FAN_PWM_COMMAND
REG_PWM_WRITE can be changed both by NCT66* itself and externally using the same ACPI mechanism)
but can also be harmful (dangerous)

SergeyMy, 2023-11-27:
//tested on nct6687
0xb00-0xc00 **PLF** (piecewise linear function)

`#define NCT6683_REG_PLF(x)   (0xb00 + (x) * 24)`

NCT6683_REG_PLF(x)+0 ... NCT6683_REG_PLF(x)+7 = pwm(x)_auto_point1_temp ... pwm(x)_auto_point7_temp
NCT6683_REG_PLF(x)+8+Y*2 = pwm(x)_auto_pointY_pwm

example:
```
0xb00: 1e 28 32 4b 4b 4b 4b 00  0c 00 3f 00 7f 00 ff 00  ff 00 ff 00 ff 00 00 05
PLF1 (30C 004)	(40C 024%)	(50C 049%)	(75C 100%)
```

Fred78290, 2023-11-28:
@SergeyMy 

I don't closed my eyes on this project :)

A little busy and submerged by email to be notified :)

SergeyMy, 2024-01-31:
pwm[]_mode 
0=DC 1=PWM
0x2f4.1 PWM3
0x2f4.2 PWM2
0x2f4.0 PWM1 May be

SergeyMy, 2024-01-31:
pwm source select (pwm[1-*]_auto_channels_temp)
20 - PCIE
01 - CPUcore (CPU)
02 - System (CPU_sock)
...
0x917 PWM2
0x91B PWM3
...

SergeyMy, 2024-01-31:
source PV (PrimaryValue for PLF) (celsium) //pwm source selec

0xca0 PLF1
0xca1 PLF2
...

SergeyMy, 2024-02-01:
[image]

closed 2024-03-04 by Fred78290