Fred78290/nct6687d #93: Access to PWM control without sudo

shvedes, 2024-03-06:
To begin with, I wanted to start with the fact that everything works fine on the MAG B550 TOMAHAWK MAX WIFI, thanks for this module!

PWM signal control works as expected, but it's a little annoying to use sudo every time. I would like to know if it is possible to control the PWM signal without sudo? I understand that this no longer applies to this driver, but to /sys in general, but I did not find the necessary information. I tried adding a regular user to the root group, but that didn't help. I have no idea how this can be implemented, or whether it is possible at all. I would be grateful for any answer, thanks!

xaizek, 2024-03-06:
`root` can write to files thanks to this line:

https://github.com/Fred78290/nct6687d/blob/0ee35ed9541bde22fe219305d1647b51ed010c5e/nct6687.c#L917

You can add `| S_IWGRP` to enable writes by members of `root` group and `| S_IWOTH` for writes by all the rest.

shvedes, 2024-03-06:
I don't quite understand what exactly you are talking about.  Can you explain more about this please?

xaizek, 2024-03-06:
I told you which changes you can do to code to have different permissions.

Alternatively, run `chmod 666 /sys/devices/platform/nct6687.2592/hwmon/hwmon*/pwm*` in some system initialization script.

shvedes, 2024-03-06:
Thanks. I will give a try

SergeyMy, 2024-03-06:
It's better this way, but!!
it is not safe for production systems

`sudo chmod 777 /sys/devices/platform/nct6*/hwmon/hwmon*/*`

xaizek, 2024-03-06:
It's also possible to do it via udev (say, in `/etc/udev/rules.d/nct6687.rules`):
```udev
SUBSYSTEM=="hwmon", DEVPATH=="/devices/platform/nct6687.*/hwmon/hwmon*", ACTION=="add", RUN+="/bin/chmod 666 %S%p/pwm*"
```
This has the advantage of being run after manual `modprobe` as well.

shvedes, 2024-03-06:
> You can add `| S_IWGRP` to enable writes by members of `root` group and `| S_IWOTH` for writes by all the rest.

So i tried as you said by changing the source code and compiling again, but it doesn't work. I also tried the udev rule and systemd service, but it still doesn't work. Only manually changing rights solves the problem. What else can I do?

Arch Linux
```
uname -r
6.7.6-zen1-1-zen
```

shvedes, 2024-03-06:
Ok, now it works with `S_IWGRP`, but not with `S_IWOTH`. 

I would like it to work as a regular user, but for now I’ll leave it that way. If you have any ideas why S_IWOTH does not work in my case, I would be glad to hear

SergeyMy, 2024-03-07:
everything works great
[image]

shvedes, 2024-03-07:
> everything works great

Doesn't work for me 😕
[image]

SergeyMy, 2024-03-07:
Maybe the previous version of the module was not unloaded from memory? but it just has to work

shvedes, 2024-03-07:
> Maybe the previous version of the module was not unloaded from memory? but it just has to work

Nope, i rebooted my PC several times. Still doesn't working

SergeyMy, 2024-03-07:
module loaded via insmod nct6687.ko or
modprobe nct6687 ?
his may be important udev can override permissions.
https://github.com/Fred78290/nct6687d/issues/93#issuecomment-1981145437

shvedes, 2024-03-07:
Loaded via modprobe, on boot as well

[image]

Installed from PKGBUILD. I just added 
`sed -i '917c\    return attr->mode | S_IWUSR | S_IWGRP | S_IWOTH;' $srcdir/nct6687d-dkms-git/nct6687.c` to package() function

SergeyMy, 2024-03-07:
`make;sudo rmmod $(lsmod|grep nct6);sudo insmod $(uname -r)/nct6*.ko;ls -la /sys/devices/platform/nct6*/hwmon/hwmon*/pwm*`
this must be done in the source code directory!,
make
rmmod
insmod
...

shvedes, 2024-03-07:
> this must be done in the source code directory!,

My bad.
[image]

SergeyMy, 2024-03-07:
I've run out of ideas, well... it could be the machinations of **selinux**. protects your safety :)

SergeyMy, 2024-03-07:
> My bad.

this is a kernel version issue
linux **kernel 6.***
:(:(:(

SergeyMy, 2024-03-07:
I was also looking for how to change the group, for example to whell :) until I learned how

shvedes, 2024-03-07:
> but using `udev` rule from above might be easier

As i said above, udev rule doesn't work for me as well for some reason

SergeyMy, 2024-03-07:
most likely for the same reason... //access for everyone should be given by root, of course and not automatically

xaizek, 2024-03-08:
> As i said above, udev rule doesn't work for me as well for some reason

I missed that when catching up.  Glob doesn't expand there and of course I tested without it...  Try corrected version:
```
SUBSYSTEM=="hwmon", DEVPATH=="/devices/platform/nct6687.*/hwmon/hwmon*", ACTION=="add", RUN+="/usr/bin/find %S%p -name pwm* -exec chmod 666 {} ;"
```

> I was also looking for how to change the group, for example to whel :) until I learned how

I think this is the API:

```c
/**
 *	sysfs_file_change_owner - change owner of a sysfs file.
 *	@kobj:	object.
 *	@name:	name of the file to change.
 *	@kuid:	new owner's kuid
 *	@kgid:	new owner's kgid
 *
 * This function looks up the sysfs entry @name under @kobj and changes the
 * ownership to @kuid/@kgid.
 *
 * Returns 0 on success or error code on failure.
 */
```

shvedes, 2024-03-08:
> Try corrected version:

It WORKS!!! Thank you so much!
[image]

shvedes, 2025-05-23:
Holy cow why this issue is still opened

closed 2025-05-23 by shvedes