Fred78290/nct6687d #96: Partially works with MSI B650M MORTAR WIFI

flippette, 2024-06-05:
Hi, thank you for your work. I have the B650M MORTAR WIFI with an NCT8867R I/O controller (found in [manual](https://download.msi.com/archive/mnu_exe/mb/MAGB650MMORTARWIFI.pdf)).

I see this when the module is loaded:

```
Jun 05 14:00:37 okinawa kernel: nct6687: Found NCT6687D or compatible chip at 0x004e:>
Jun 05 14:00:37 okinawa kernel: nct6687 nct6687.2592: NCT6687D EC firmware version 0.>
Jun 05 14:00:37 okinawa systemd-modules-load[547]: Inserted module 'nct6687'
```

This is what `sensors` outputs:

```
nct6687-isa-0a20
Adapter: ISA adapter
+12V:           12.10 V  (min = +12.10 V, max = +12.10 V)
+5V:             5.03 V  (min =  +5.03 V, max =  +5.04 V)
+3.3V:           3.36 V  (min =  +3.36 V, max =  +3.36 V)
CPU Soc:         1.29 V  (min =  +1.29 V, max =  +1.30 V)
CPU Vcore:     816.00 mV (min =  +0.75 V, max =  +1.42 V)
CPU 1P8:         0.00 V  (min =  +0.00 V, max =  +0.00 V)
CPU VDDP:        0.00 V  (min =  +0.00 V, max =  +0.00 V)
DRAM:            1.34 V  (min =  +1.34 V, max =  +1.35 V)
Chipset:       316.00 mV (min =  +0.29 V, max =  +0.35 V)
CPU SA:        580.00 mV (min =  +0.58 V, max =  +0.61 V)
Voltage #2:      1.54 V  (min =  +1.54 V, max =  +1.54 V)
AVCC3:           3.31 V  (min =  +3.31 V, max =  +3.31 V)
AVSB:            3.36 V  (min =  +3.36 V, max =  +3.36 V)
VBat:            2.05 V  (min =  +2.05 V, max =  +2.05 V)
CPU Fan:        485 RPM  (min =  485 RPM, max = 1059 RPM)
Pump Fan:      3069 RPM  (min = 3061 RPM, max = 3076 RPM)
System Fan #1:  564 RPM  (min =  558 RPM, max =  710 RPM)
System Fan #2: 1268 RPM  (min = 1226 RPM, max = 1268 RPM)
System Fan #3:  558 RPM  (min =  543 RPM, max =  710 RPM)
System Fan #4:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #5:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #6:    0 RPM  (min =    0 RPM, max =    0 RPM)
CPU:            +53.0°C  (low  = +53.0°C, high = +71.0°C)
System:         +49.5°C  (low  = +47.0°C, high = +49.5°C)
VRM MOS:        +42.0°C  (low  = +42.0°C, high = +44.0°C)
PCH:            +51.5°C  (low  = +49.0°C, high = +51.5°C)
CPU Socket:     +43.0°C  (low  = +42.5°C, high = +45.0°C)
PCIe x1:        +80.0°C  (low  = +78.5°C, high = +80.0°C)
M2_1:            +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
```

`M2_1`, `CPU 1P8` and `CPU VDDP` doesn't seem to work, `PCIe x1` seems really high to me (is it?), this board doesn't have fan headers 4-6.

My hardware:
- MSI B650M MORTAR WIFI
- Ryzen 9 7950X
- Both M.2 slots have SSDs
- All fan headers on the board are used
- I don't know what `PCIe x1` is referring to, this board only goes down to a x4 slot

mahmoudhossam, 2024-11-03:
The PCIe x1 temperature has been abnormally high on some boards for a while, not an issue with this particular kernel module https://www.reddit.com/r/buildapc/comments/101upqu/pcie_x1_temperature_abnormally_high/

vladns, 2025-01-06:
On this motherboard I get errors:
```
nct6687-isa-0a20
Adapter: ISA adapter
+12V:           12.34 V  (min = +12.22 V, max = +12.36 V)
+5V:             4.96 V  (min =  +4.96 V, max =  +4.99 V)
+3.3V:           3.33 V  (min =  +3.32 V, max =  +3.33 V)
CPU Soc:         1.20 V  (min =  +1.20 V, max =  +1.21 V)
CPU Vcore:     782.00 mV (min =  +0.73 V, max =  +1.37 V)
CPU 1P8:         0.00 V  (min =  +0.00 V, max =  +0.00 V)
CPU VDDP:        0.00 V  (min =  +0.00 V, max =  +0.00 V)
DRAM:            1.10 V  (min =  +1.10 V, max =  +1.10 V)
Chipset:       328.00 mV (min =  +0.29 V, max =  +0.38 V)
CPU SA:        552.00 mV (min =  +0.51 V, max =  +0.64 V)
Voltage #2:      1.54 V  (min =  +1.53 V, max =  +1.54 V)
AVCC3:           3.34 V  (min =  +3.33 V, max =  +3.35 V)
AVSB:            3.33 V  (min =  +3.32 V, max =  +3.34 V)
VBat:            2.04 V  (min =  +2.04 V, max =  +2.04 V)
CPU Fan:       1485 RPM  (min = 1311 RPM)
Pump Fan:         0 RPM  (min =    0 RPM)
System Fan #1: 1147 RPM  (min = 1108 RPM)
System Fan #2: 1219 RPM  (min = 1097 RPM)
System Fan #3:    0 RPM  (min =    0 RPM)
System Fan #4:    0 RPM  (min =    0 RPM)
System Fan #5:    0 RPM  (min =    0 RPM)
System Fan #6:    0 RPM  (min =    0 RPM)
ERROR: Can't get value of subfeature fan9_input: Kernel interface error
ERROR: Can't get value of subfeature fan9_min: Kernel interface error
fan9:               N/A  (min =    0 RPM)
ERROR: Can't get value of subfeature fan10_input: Kernel interface error
ERROR: Can't get value of subfeature fan10_min: Kernel interface error
fan10:              N/A  (min =    0 RPM)
ERROR: Can't get value of subfeature temp1_max_hyst: Kernel interface error
ERROR: Can't get value of subfeature temp1_crit: Kernel interface error
CPU:            +65.0°C  (low  = +57.0°C)
                         (high = +78.0°C, hyst =  +0.0°C)
ERROR: Can't get value of subfeature temp1_type: Kernel interface error
                         (crit =  +0.0°C)  sensor = disabled
ERROR: Can't get value of subfeature temp2_max_hyst: Kernel interface error
ERROR: Can't get value of subfeature temp2_crit: Kernel interface error
System:         +48.0°C  (low  = +43.0°C)
                         (high = +51.0°C, hyst =  +0.0°C)
ERROR: Can't get value of subfeature temp2_type: Kernel interface error
                         (crit =  +0.0°C)  sensor = disabled
ERROR: Can't get value of subfeature temp3_max_hyst: Kernel interface error
ERROR: Can't get value of subfeature temp3_crit: Kernel interface error
VRM MOS:        +43.5°C  (low  = +37.5°C)
                         (high = +45.5°C, hyst =  +0.0°C)
ERROR: Can't get value of subfeature temp3_type: Kernel interface error
                         (crit =  +0.0°C)  sensor = disabled
ERROR: Can't get value of subfeature temp4_max_hyst: Kernel interface error
ERROR: Can't get value of subfeature temp4_crit: Kernel interface error
PCH:            +53.5°C  (low  = +46.0°C)
                         (high = +57.0°C, hyst =  +0.0°C)
ERROR: Can't get value of subfeature temp4_type: Kernel interface error
                         (crit =  +0.0°C)  sensor = disabled
ERROR: Can't get value of subfeature temp5_max_hyst: Kernel interface error
ERROR: Can't get value of subfeature temp5_crit: Kernel interface error
CPU Socket:     +45.0°C  (low  = +38.0°C)
                         (high = +52.5°C, hyst =  +0.0°C)
ERROR: Can't get value of subfeature temp5_type: Kernel interface error
                         (crit =  +0.0°C)  sensor = disabled
ERROR: Can't get value of subfeature temp6_max_hyst: Kernel interface error
ERROR: Can't get value of subfeature temp6_crit: Kernel interface error
PCIe x1:        +76.5°C  (low  = +75.0°C)
                         (high = +80.5°C, hyst =  +0.0°C)
ERROR: Can't get value of subfeature temp6_type: Kernel interface error
                         (crit =  +0.0°C)  sensor = disabled
```
Linux kernel 6.12.8

flippette, 2025-01-06:
@vladns You may need to blacklist the `nct6683` module.