Fred78290/nct6687d #97: Weird results on MSI B550 gaming gen3

waanito, 2024-06-22:
Hello, 
thank you for your work.
 I have the B550gaming gen3
(https://download.msi.com/archive/mnu_exe/mb/B550GAMINGGEN3.pdf)

I see this when the module is loaded:
```
~$ sudo dmesg |grep nct6687
[    4.650787] nct6687: loading out-of-tree module taints kernel.
[    4.651186] nct6687: module verification failed: signature and/or required key missing - tainting kernel
[    5.679770] nct6687: Found NCT6687D or compatible chip at 0x004e:0x0a20
[    5.681634] nct6687 nct6687.2592: NCT6687D EC firmware version 255.255 build 00/00/00
```
and these weird results:
```
~$ sudo sensors
nct6687-isa-0a20
Adapter: ISA adapter
+12V:          -16.58 V  (min = -16.58 V, max = -16.58 V)
+5V:           240.00 mV (min =  +0.24 V, max =  +0.24 V)
+3.3V:         752.00 mV (min =  +0.40 V, max =  +0.75 V)
CPU Soc:        15.00 mV (min =  +0.01 V, max =  +0.01 V)
CPU Vcore:       4.09 V  (min =  +4.09 V, max =  +4.09 V)
CPU 1P8:       368.00 mV (min =  +0.37 V, max =  +0.37 V)
CPU VDDP:        4.08 V  (min =  +4.08 V, max =  +4.08 V)
DRAM:            8.19 V  (min =  +8.19 V, max =  +8.19 V)
Chipset:         4.09 V  (min =  +4.09 V, max =  +4.09 V)
CPU SA:          4.09 V  (min =  +4.09 V, max =  +4.09 V)
Voltage #2:    368.00 mV (min =  +0.37 V, max =  +0.69 V)
AVCC3:           4.08 V  (min =  +4.08 V, max =  +4.08 V)
AVSB:            4.08 V  (min =  +4.08 V, max =  +4.08 V)
VBat:            4.08 V  (min =  +4.08 V, max =  +4.08 V)
CPU Fan:       65311 RPM  (min = 65311 RPM, max = 65311 RPM)
Pump Fan:      65311 RPM  (min = 65311 RPM, max = 65311 RPM)
System Fan #1:  514 RPM  (min =  514 RPM, max =  514 RPM)
System Fan #2:  514 RPM  (min =  514 RPM, max =  514 RPM)
System Fan #3:  514 RPM  (min =  514 RPM, max =  514 RPM)
System Fan #4: 65311 RPM  (min = 65311 RPM, max = 65311 RPM)
System Fan #5: 65311 RPM  (min = 65311 RPM, max = 65311 RPM)
System Fan #6: 1538 RPM  (min = 1538 RPM, max = 1538 RPM)
CPU:             +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
System:          +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
VRM MOS:         +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
PCH:             +0.5°C  (low  =  +0.0°C, high =  +0.5°C)
CPU Socket:    +255.5°C  (low  = +255.0°C, high = +255.5°C)
PCIe x1:       +255.5°C  (low  = +255.0°C, high = +255.5°C)
M2_1:          +255.5°C  (low  = +255.0°C, high = +255.5°C)

nct6797-isa-0a20
Adapter: ISA adapter
in0:                   856.00 mV (min =  +0.00 V, max =  +1.74 V)
in1:                     1.04 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                     3.36 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                     3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                     1.06 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                     2.04 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                   920.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                     3.36 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                     3.68 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                     1.84 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                  920.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                  712.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                  808.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                  600.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                    1.56 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     0 RPM  (min =    0 RPM)
fan2:                  1812 RPM  (min =    0 RPM)
fan3:                     0 RPM  (min =    0 RPM)
fan4:                   890 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
fan6:                  1819 RPM  (min =    0 RPM)
SYSTIN:                 +18.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = CPU diode
CPUTIN:                 +29.0°C  (high = +125.0°C, hyst = +90.0°C)  sensor = thermistor
AUXTIN0:                +30.5°C  (high = +125.0°C, hyst = +90.0°C)  sensor = thermistor
AUXTIN1:                +30.0°C    sensor = thermistor
AUXTIN2:                +42.0°C    sensor = thermistor
AUXTIN3:                 -4.0°C    sensor = thermistor
SMBUSMASTER 0:          +45.0°C
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C
PCH_CHIP_TEMP:           +0.0°C
PCH_CPU_TEMP:            +0.0°C
PCH_MCH_TEMP:            +0.0°C
Agent0 Dimm0:            +0.0°C
TSI0_TEMP:              +45.2°C
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           disabled

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +32.6°C
```

Hardware:
MSI B550 gaming gen3
AMD Ryzen 5 5500
NVIDIA Corporation GK208B [GeForce GT 710]
3 SATA HDD
1 SATA SSD
2 sysfans

Fred78290, 2024-06-23:
Hello, probably the chipset is not a nct6687 but a nct6775.

To check it use sensors-detect tool.

waanito, 2024-06-24:
Hello,
thanks. unloading nst6687 and re-running sensors-detect it now uses built in  nct6775
```
Found `Nuvoton NCT6797D Super IO Sensors'                   Success!
    (address 0xa20, driver `nct6775')
```
which has better results but not quite there.
I'll keep at it and report back here for others benefit  once I solve. Thanks for all your help.
```
jonny@tootee:~$ sensors
nct6797-isa-0a20
Adapter: ISA adapter
in0:                   256.00 mV (min =  +0.00 V, max =  +1.74 V)
in1:                     1.04 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                     3.36 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                     3.39 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                     1.06 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                     2.04 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                   952.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                     3.36 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                     3.66 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                     1.84 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                  920.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                  688.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                  784.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                  600.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                    1.56 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     0 RPM  (min =    0 RPM)
fan2:                  1812 RPM  (min =    0 RPM)
fan3:                     0 RPM  (min =    0 RPM)
fan4:                  1708 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
fan6:                  1819 RPM  (min =    0 RPM)
SYSTIN:                 +18.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = CPU diode
CPUTIN:                 +29.0°C  (high = +125.0°C, hyst = +90.0°C)  sensor = thermistor
AUXTIN0:                +28.5°C  (high = +125.0°C, hyst = +90.0°C)  sensor = thermistor
AUXTIN1:                +31.0°C    sensor = thermistor
AUXTIN2:                +43.0°C    sensor = thermistor
AUXTIN3:                 -4.0°C    sensor = thermistor
SMBUSMASTER 0:          +44.0°C
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C
PCH_CHIP_TEMP:           +0.0°C
PCH_CPU_TEMP:            +0.0°C
PCH_MCH_TEMP:            +0.0°C
Agent0 Dimm0:            +0.0°C
TSI0_TEMP:              +44.0°C
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           disabled

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +32.2°C
```

Fred78290, 2024-06-24:
`which has better results but not quite there`

I know, i have a MB with this chipset.

closed 2024-06-24 by Fred78290