Fred78290/nct6687d #102: The driver does not compile with Debian Sid kernel 6.11.2

Makhno77, 2024-10-07:
As per the title, the driver does not compile: I attach the make.log
root@debian:~# cat /var/lib/dkms/nct6687d/1/build/make.log
DKMS make.log for nct6687d-1 for kernel 6.11.2-amd64 (x86_64)
lun 7 ott 2024, 23:22:24, CEST
fatal: .git non è un repository Git (né lo è alcuna delle directory genitrici)
fatal: .git non è un repository Git (né lo è alcuna delle directory genitrici)
make -C /lib/modules/6.11.2-amd64/build M=/var/lib/dkms/nct6687d/1/build modules
make[1]: attenzione: jobserver non disponibile, viene usato -j1. Aggiungere «+» alla regola make superiore.
make[1]: ingresso nella directory «/usr/src/linux-headers-6.11.2-amd64»
fatal: .git non è un repository Git (né lo è alcuna delle directory genitrici)
fatal: .git non è un repository Git (né lo è alcuna delle directory genitrici)
  CC [M]  /var/lib/dkms/nct6687d/1/build/nct6687.o
/var/lib/dkms/nct6687d/1/build/nct6687.c:40:9: warning: "MIN" redefined
   40 | #define MIN(a,b) (((a)<(b))?(a):(b))
      |         ^~~
In file included from /usr/src/linux-headers-6.11.2-common/include/linux/ioport.h:15,
                 from /usr/src/linux-headers-6.11.2-common/include/linux/acpi.h:12,
                 from /var/lib/dkms/nct6687d/1/build/nct6687.c:27:
/usr/src/linux-headers-6.11.2-common/include/linux/minmax.h:329:9: note: this is the location of the previous definition
  329 | #define MIN(a,b) __cmp(min,a,b)
      |         ^~~
/var/lib/dkms/nct6687d/1/build/nct6687.c:41:9: warning: "MAX" redefined
   41 | #define MAX(a,b) (((a)>(b))?(a):(b))
      |         ^~~
/usr/src/linux-headers-6.11.2-common/include/linux/minmax.h:330:9: note: this is the location of the previous definition
  330 | #define MAX(a,b) __cmp(max,a,b)
      |         ^~~
/var/lib/dkms/nct6687d/1/build/nct6687.c:1087:19: error: initialization of ‘void (*)(struct platform_device *)’ from incompatible pointer type ‘int (*)(struct platform_device *)’ [-Wincompatible-pointer-types]
 1087 |         .remove = nct6687_remove,
      |                   ^~~~~~~~~~~~~~
/var/lib/dkms/nct6687d/1/build/nct6687.c:1087:19: note: (near initialization for ‘nct6687_driver.<anonymous>.remove’)
make[3]: *** [/usr/src/linux-headers-6.11.2-common/scripts/Makefile.build:249: /var/lib/dkms/nct6687d/1/build/nct6687.o] Errore 1
make[2]: *** [/usr/src/linux-headers-6.11.2-common/Makefile:1951: /var/lib/dkms/nct6687d/1/build] Errore 2
make[1]: *** [/usr/src/linux-headers-6.11.2-common/Makefile:236: __sub-make] Errore 2
make[1]: uscita dalla directory «/usr/src/linux-headers-6.11.2-amd64»
make: *** [Makefile:46: dkms/build] Errore 2
root@debian:~#

Fred78290, 2024-10-08:
Get latest version and rebuild the dkms package. It was fixed last month

Makhno77, 2024-10-08:
Thanks for the reply but the error seems the same
[6687b.log](https://github.com/user-attachments/files/17292119/6687b.log)

Makhno77, 2024-10-08:
In data martedì 8 ottobre 2024 15:21:51 CEST, xaizek ha scritto:
> You didn't update the sources or didn't run `make dkms/install`

You're right, the second time I just ran make install, too much haste. 
Sorry for the noise.
Cheers

-- 
Marco Valli

Fred78290, 2024-10-08:
@xaizek Thanks

closed 2024-10-08 by Fred78290