Fred78290/nct6687d #103: Works on ASRock LiveMixer B650

DaGhostman, 2024-10-17:
It seems to be working better than the default nct6683 driver!

Although some of the voltages don't seem to be detected(calculated?) properly?

```
nct6686-isa-0a20
Adapter: ISA adapter
+12V:            0.00 V  (min =  +0.00 V, max =  +0.00 V)
+5V:             0.00 V  (min =  +0.00 V, max =  +0.00 V)
+3.3V:           0.00 V  (min =  +0.00 V, max =  +3.42 V)
CPU Soc:         0.00 V  (min =  +0.00 V, max =  +0.00 V)
CPU Vcore:       0.00 V  (min =  +0.00 V, max =  +0.00 V)
CPU 1P8:         0.00 V  (min =  +0.00 V, max =  +0.00 V)
CPU VDDP:        0.00 V  (min =  +0.00 V, max =  +0.00 V)
DRAM:            0.00 V  (min =  +0.00 V, max =  +0.00 V)
Chipset:         0.00 V  (min =  +0.00 V, max =  +0.00 V)
CPU SA:          0.00 V  (min =  +0.00 V, max =  +0.00 V)
Voltage #2:      0.00 V  (min =  +0.00 V, max =  +0.00 V)
AVCC3:           0.00 V  (min =  +0.00 V, max =  +0.00 V)
AVSB:            0.00 V  (min =  +0.00 V, max =  +3.42 V)
VBat:            0.00 V  (min =  +0.00 V, max =  +1.81 V)
CPU Fan:        783 RPM  (min =  760 RPM, max = 1392 RPM)
Pump Fan:       904 RPM  (min =  875 RPM, max = 1595 RPM)
System Fan #1: 1062 RPM  (min = 1047 RPM, max = 1562 RPM)
System Fan #2:  907 RPM  (min =  870 RPM, max = 1591 RPM)
System Fan #3: 1061 RPM  (min = 1036 RPM, max = 1566 RPM)
System Fan #4: 1685 RPM  (min = 1623 RPM, max = 1746 RPM)
System Fan #5:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #6:    0 RPM  (min =    0 RPM, max =    0 RPM)
CPU:            +11.5°C  (low  =  +5.5°C, high = +33.5°C)
System:         +31.5°C  (low  = +23.0°C, high = +59.0°C)
VRM MOS:        +30.5°C  (low  = +24.0°C, high = +49.5°C)
PCH:            +41.0°C  (low  = +35.0°C, high = +74.0°C)
CPU Socket:     +70.0°C  (low  = +59.0°C, high = +80.0°C)
PCIe x1:        +62.0°C  (low  = +61.0°C, high = +62.0°C)
M2_1:           +63.0°C  (low  = +61.0°C, high = +63.0°C)
```

cavity-dump, 2024-12-30:
I got myself this board and noticed the same. After some investigation I figured out the correct registers which work in my own cobbled together program.

Motherboard : Asrock B650 Livemixer
SIO chip : Nuvoton NCT6686D
Chip ID : 0xD441
Address found : 0x002E:0x0A20
Voltage registers start at 0x108 instead of 0x120. (NCT6687_REG_VOLTAGE)
in order :
 * Vcore Voltage
 * +5.0V (multiplier 5.02)
 * +12.0V (multiplier 12)
 * ignore
 * VDDCR_SOC_Voltage
 * VDD_MISC Voltage
 * Dram Voltage
 * +3.3V
Note : names and multipliers taken from Asrock windows software as they are conveniently included as a separate XML file.

Temp registers start at 0x100 (NCT6687_REG_TEMP)
in order :
*M/B temp (?)
*CPU socket (?)
*CPU temp
I'm a bit unsure as the windows software only shows 2 temperatures

Now, how to integrate this into this driver I'm not sure. Hopefully the author can help here.

Fred78290, 2024-12-31:
@cavity-dump @DaGhostman 

The nct6686 module is already present in standard linux distro.

cavity-dump, 2024-12-31:
You mean the nct6683 module?
Thanks! I could not get that one to work before, but now I found out that it indeed does work by using modprobe nct6683 force=on

I'm a bit confused about the name and scope of these modules. Perhaps you can share some light on how this should work ?

* the nct6683 module works (temp + fans OK) on a nct6686d chip, but only if forced
* this nct6687 module works (no forcing needed, fans shown correctly), but the voltages are in the wrong place

Both don't have nct6686 in the module name, but both can detect it and list it as a nct6686. However both don't work (out of the box).

DaGhostman, 2024-12-31:
Also on nct6683 the PWM control for the fans is not working, while with the nct6687d the PWM is manageable, right now I see all fan control files as read-only.