Fred78290/nct6687d #107: No fan RPM on some fans on MSI X870 Tomahawk

nfp0, 2024-10-25:
This module seems to have exposed a lot of values for my MSI X870 Tomahawk motherboard, but some seem to be incorrect.

For example, the CPU Fan RPM is reading correctly, but the System Fan 2 and System Fan 3 should be reporting some RPM value, as I've confirmed they are rotating and the values are reading correctly on the BIOS.
A few temperature sensors also seem to be missing.

Here's the output of `sensors` after doing a `sensors-detect`:
```
nct6687-isa-0a20
Adapter: ISA adapter
+12V:           12.14 V  (min = +12.14 V, max = +12.17 V)
+5V:             5.06 V  (min =  +5.06 V, max =  +5.06 V)
+3.3V:           3.36 V  (min =  +0.00 V, max =  +3.36 V)
CPU Soc:         1.21 V  (min =  +1.21 V, max =  +1.22 V)
CPU Vcore:       1.09 V  (min =  +1.04 V, max =  +1.09 V)
CPU 1P8:         3.36 V  (min =  +3.36 V, max =  +3.36 V)
CPU VDDP:        3.36 V  (min =  +3.36 V, max =  +3.36 V)
DRAM:            1.41 V  (min =  +1.41 V, max =  +1.41 V)
Chipset:       332.00 mV (min =  +0.32 V, max =  +0.34 V)
CPU SA:        702.00 mV (min =  +0.70 V, max =  +0.71 V)
Voltage #2:      1.52 V  (min =  +1.52 V, max =  +1.52 V)
AVCC3:           3.30 V  (min =  +3.30 V, max =  +3.30 V)
AVSB:            3.36 V  (min =  +0.00 V, max =  +3.36 V)
VBat:            2.04 V  (min =  +0.00 V, max =  +2.04 V)
CPU Fan:        414 RPM  (min =  409 RPM, max =  414 RPM)
Pump Fan:         0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #1:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #2:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #3:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #4:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #5:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #6:    0 RPM  (min =    0 RPM, max =    0 RPM)
CPU:            +49.0°C  (low  = +48.0°C, high = +50.0°C)
System:         +45.0°C  (low  = +45.0°C, high = +45.0°C)
VRM MOS:        +46.0°C  (low  = +45.0°C, high = +46.0°C)
PCH:            +43.0°C  (low  = +42.0°C, high = +43.0°C)
CPU Socket:      +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
PCIe x1:         +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
M2_1:            +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
```

TheCompGeek, 2024-11-03:
Can confirm as well on an MSI MPG X870E CARBON WIFI board as well:
```
nct6687-isa-0a20
Adapter: ISA adapter
+12V:           12.29 V  (min = +12.26 V, max = +12.29 V)
+5V:             5.04 V  (min =  +5.04 V, max =  +5.04 V)
+3.3V:           3.34 V  (min =  +3.34 V, max =  +3.34 V)
CPU Soc:         1.31 V  (min =  +1.31 V, max =  +1.31 V)
CPU Vcore:       1.30 V  (min =  +1.30 V, max =  +1.30 V)
CPU 1P8:         3.34 V  (min =  +3.34 V, max =  +3.34 V)
CPU VDDP:        3.34 V  (min =  +3.34 V, max =  +3.34 V)
DRAM:            1.37 V  (min =  +1.37 V, max =  +1.37 V)
Chipset:         2.04 V  (min =  +2.04 V, max =  +2.04 V)
CPU SA:        422.00 mV (min =  +0.42 V, max =  +0.42 V)
Voltage #2:      1.54 V  (min =  +1.54 V, max =  +1.54 V)
AVCC3:           3.30 V  (min =  +3.30 V, max =  +3.30 V)
AVSB:            3.34 V  (min =  +3.34 V, max =  +3.34 V)
VBat:            2.04 V  (min =  +2.04 V, max =  +2.04 V)
CPU Fan:       1321 RPM  (min = 1320 RPM, max = 1321 RPM)
Pump Fan:         0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #1:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #2:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #3:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #4:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #5:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #6:    0 RPM  (min =    0 RPM, max =    0 RPM)
CPU:            +47.0°C  (low  = +47.0°C, high = +47.0°C)
System:         +49.5°C  (low  = +49.0°C, high = +49.5°C)
VRM MOS:        +38.5°C  (low  = +38.0°C, high = +38.5°C)
PCH:            +48.5°C  (low  = +48.5°C, high = +49.0°C)
CPU Socket:      +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
PCIe x1:        +47.0°C  (low  = +47.0°C, high = +47.0°C)
M2_1:          +216.0°C  (low  = +216.0°C, high = +216.0°C)

```

f8lxninja, 2024-11-25:
Confirmed on MSI PRO X870-P WIFI:

```
nct6687-isa-0a20
Adapter: ISA adapter
+12V:           12.22 V  (min = +12.19 V, max = +12.24 V)
+5V:             5.01 V  (min =  +4.98 V, max =  +5.02 V)
+3.3V:           3.34 V  (min =  +2.04 V, max =  +3.34 V)
CPU Soc:         1.31 V  (min =  +1.31 V, max =  +1.31 V)
CPU Vcore:       1.34 V  (min =  +1.32 V, max =  +1.35 V)
CPU 1P8:         3.34 V  (min =  +3.34 V, max =  +3.34 V)
CPU VDDP:        3.34 V  (min =  +3.33 V, max =  +3.34 V)
DRAM:            1.41 V  (min =  +1.41 V, max =  +1.42 V)
Chipset:       346.00 mV (min =  +0.33 V, max =  +0.35 V)
CPU SA:        778.00 mV (min =  +0.78 V, max =  +0.80 V)
Voltage #2:      1.53 V  (min =  +1.53 V, max =  +1.53 V)
AVCC3:           3.32 V  (min =  +3.32 V, max =  +3.33 V)
AVSB:            3.34 V  (min =  +0.00 V, max =  +3.34 V)
VBat:            2.04 V  (min =  +0.00 V, max =  +2.04 V)
CPU Fan:       1612 RPM  (min =    0 RPM, max = 2090 RPM)
Pump Fan:         0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #1:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #2:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #3:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #4:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #5:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #6:    0 RPM  (min =    0 RPM, max =    0 RPM)
CPU:            +53.0°C  (low  = +50.0°C, high = +67.0°C)
System:         +42.5°C  (low  = +41.0°C, high = +42.5°C)
VRM MOS:        +39.5°C  (low  = +38.5°C, high = +41.0°C)
PCH:            +38.5°C  (low  = +37.0°C, high = +38.5°C)
CPU Socket:      +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
PCIe x1:         +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
M2_1:            +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
```

demomanca, 2024-12-19:
Some of us are attempting to resolve this in the LibreHardwareMonitor project. It appears MSI has used some different pinouts on this motherboard than other implementations, and it doesn't appear to be as simple as "remapping them". Any assistance on this would be appreciated, as it seems to be effecting all the newer MSI boards using this chip (X870 and Z890)

Fred78290, 2024-12-19:
@demomanca I don't have this motherboard.

demomanca, 2024-12-19:
> @demomanca I don't have this motherboard.

Sorry, I should have been clearer, it wasn't so much about that particular board, it was more about how you went about finding the registers etc, or is it a case of (as we have been doing) just trial, error, poke and prod.

sgoth, 2025-01-29:
I see the same for MSI B850 Tomahawk:

```
nct6687-isa-0a20
Adapter: ISA adapter
+12V:           12.12 V  (min = +12.10 V, max = +12.12 V)
+5V:             5.06 V  (min =  +5.05 V, max =  +5.06 V)
+3.3V:           3.35 V  (min =  +0.00 V, max =  +3.35 V)
CPU Soc:         1.21 V  (min =  +1.21 V, max =  +1.21 V)
CPU Vcore:       1.22 V  (min =  +1.21 V, max =  +1.23 V)
CPU 1P8:         3.35 V  (min =  +3.34 V, max =  +3.35 V)
CPU VDDP:        3.35 V  (min =  +3.34 V, max =  +3.35 V)
DRAM:            1.36 V  (min =  +1.36 V, max =  +1.36 V)
Chipset:       320.00 mV (min =  +0.31 V, max =  +0.33 V)
CPU SA:        764.00 mV (min =  +0.76 V, max =  +0.77 V)
Voltage #2:      1.54 V  (min =  +1.54 V, max =  +1.54 V)
AVCC3:           3.33 V  (min =  +3.32 V, max =  +3.34 V)
AVSB:            3.35 V  (min =  +0.00 V, max =  +3.35 V)
VBat:            2.05 V  (min =  +0.00 V, max =  +2.05 V)
CPU Fan:        404 RPM  (min =  399 RPM, max =  459 RPM)
Pump Fan:         0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #1:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #2:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #3:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #4:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #5:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #6:    0 RPM  (min =    0 RPM, max =    0 RPM)
CPU:            +43.0°C  (low  = +43.0°C, high = +50.0°C)
System:         +38.5°C  (low  = +38.5°C, high = +41.5°C)
VRM MOS:        +38.5°C  (low  = +38.0°C, high = +39.0°C)
PCH:            +39.0°C  (low  = +39.0°C, high = +39.5°C)
CPU Socket:      +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
PCIe x1:         +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
M2_1:            +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
```

Fred78290, 2025-01-29:
I don't have the ability to develop on theses motherboard
so the repository is collaborative.

Ncl8, 2025-02-01:
I don't know if this is of any help but people have been trying to get X870(E) / NCT6687-R / NCT668D chips working for LibreHardwareMonitor.

https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/1488
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/1490
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/1513

and a fork of [LibreHardwareMonitor](https://github.com/Alcolawl/LibreHardwareMonitor) where they have got these chips to work at least partially.

demomanca, 2025-02-06:
Hi @Fred78290 ,

Pull request for LHM is here https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1621

You can see what we had to do to get those other fan headers working. Not sure how we manage that here however. Is there a way to define "if this board, then this layout" with this module?

Fred78290, 2025-02-07:
@demomanca 

Thx, I read your changes and I have just changed the number of fans to 16.

I created a branch named **16-fans**, could you check on your motherboard that all fans appear?

Before install, remove the running module with rmmod nct6687 followed by make dkms/clean; make dkms/install

[image]

demomanca, 2025-02-11:
Thanks @Fred78290 , I'll try and give this a go this week and see if it works.

SCjona, 2025-02-12:
I have a MSI X870E Carbon with the same issue. I built the module with the `16-fans` branch and now my system fans are listed. System Fan #14 is sys_fan_1 header on the mainboard and System Fan #13 is the sys_fan_2 header on the mainboard.

Fred78290, 2025-02-12:
@SCjona 

Could try by changing in file nct6687d.c:1665

```
#define NCT6687_REG_PWM(x) ((x) < 8 ? (0x160 + (x)) : (0xE00 + (x)))
```

inplace of

```
#define NCT6687_REG_PWM(x) (0x160 + (x))
```

SCjona, 2025-02-12:
@Fred78290 that change seems to help, but writing to the pwm_enable values still has no effect. also the fan number and pwm number probably isn't a 1:1 match (i assume those with a pwm value are valid fans in my system as there are 8 non zero values and 8 fan headers on my mainboard)

<details>
<summary> PWM values </summary>

```
pwm1:41
pwm10:184
pwm10_enable:99
pwm11:184
pwm11_enable:99
pwm12:184
pwm12_enable:99
pwm13:184
pwm13_enable:99
pwm14:184
pwm14_enable:99
pwm15:0
pwm15_enable:99
pwm16:0
pwm16_enable:99
pwm1_enable:99
pwm2:49
pwm2_enable:99
pwm3:0
pwm3_enable:99
pwm4:153
pwm4_enable:99
pwm5:0
pwm5_enable:99
pwm6:0
pwm6_enable:99
pwm7:0
pwm7_enable:99
pwm8:0
pwm8_enable:99
pwm9:0
pwm9_enable:99
```

</details>

demomanca, 2025-02-12:
> PWM control does not seem to work however for the system fans. it does work for fan 1 and 2 (cpu fan and pump). for the system fans I get `cat pwm15_enabled => 0`, writing to it has no effect. I also tried changing the `NCT6687_NUM_REG_PWM` constant to 16, but that did not help.

Yeah, we haven't worked out the PWM control signal points for those fans yet. MSI and Nuvoton won't tell us where the register is. :(

sgoth, 2025-02-15:
> `System Fan #14` is sys_fan_1 header on the mainboard

Can confirm that working for MSI B850 Tomahawk too

dyfrgi, 2025-03-17:
The above referenced LibreHardwareMonitor PR has the reverse engineered register locations in it. They're now storing the full curve in registers and, to set the "current" speed, you must set all of them. There's further discussion and description in https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/1513, though it's a bit hard to find in the huge thread. So fixing this should now just be a matter of implementation, so long as we trust the LHM PR.

aamaanaa, 2025-04-22:
same issue here, als i can not see `T_SENSOR`

```
r8169_0_a00:00-mdio-0
Adapter: MDIO adapter
temp1:        +53.0°C  (high = +110.0°C)

r8169_0_900:00-mdio-0
Adapter: MDIO adapter
temp1:        +49.5°C  (high = +120.0°C)

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +47.9°C  

nvme-pci-0500
Adapter: PCI adapter
Composite:    +48.9°C  (low  =  -0.1°C, high = +84.8°C)
                       (crit = +94.8°C)

nvme-pci-0400
Adapter: PCI adapter
Composite:    +39.9°C  (low  = -273.1°C, high = +89.8°C)
                       (crit = +94.8°C)
Sensor 1:     +39.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +32.9°C  (low  = -273.1°C, high = +65261.8°C)

nct6687-isa-0a20
Adapter: ISA adapter
+12V:           11.98 V  (min = +11.95 V, max = +12.00 V)
+5V:             5.06 V  (min =  +5.04 V, max =  +5.07 V)
+3.3V:           3.34 V  (min =  +0.00 V, max =  +3.34 V)
CPU Soc:         1.21 V  (min =  +1.20 V, max =  +1.21 V)
CPU Vcore:       1.29 V  (min =  +1.09 V, max =  +1.33 V)
CPU 1P8:         3.34 V  (min =  +3.34 V, max =  +3.34 V)
CPU VDDP:        3.34 V  (min =  +3.34 V, max =  +3.34 V)
DRAM:            1.37 V  (min =  +1.37 V, max =  +1.37 V)
Chipset:         1.10 V  (min =  +1.10 V, max =  +1.11 V)
CPU SA:        348.00 mV (min =  +0.34 V, max =  +0.36 V)
Voltage #2:      1.54 V  (min =  +1.53 V, max =  +1.54 V)
AVCC3:           3.34 V  (min =  +3.31 V, max =  +3.34 V)
AVSB:            3.34 V  (min =  +0.00 V, max =  +3.34 V)
VBat:            2.05 V  (min =  +0.00 V, max =  +2.05 V)
CPU Fan:          0 RPM  (min =    0 RPM, max =    0 RPM)
Pump Fan:      3625 RPM  (min = 3539 RPM, max = 3658 RPM)
System Fan #1:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #2:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #3:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #4:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #5:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #6:    0 RPM  (min =    0 RPM, max =    0 RPM)
CPU:            +47.0°C  (low  = +45.0°C, high = +70.0°C)
System:         +48.5°C  (low  = +48.0°C, high = +49.0°C)
VRM MOS:        +55.5°C  (low  = +53.0°C, high = +59.5°C)
PCH:            +50.0°C  (low  = +49.0°C, high = +50.0°C)
CPU Socket:      +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
PCIe x1:        +49.0°C  (low  = +48.0°C, high = +49.0°C)
M2_1:           +20.5°C  (low  = +20.0°C, high = +20.5°C)

spd5118-i2c-1-53
Adapter: SMBus PIIX4 adapter port 0 at 0b00
temp1:        +35.5°C  (low  =  +0.0°C, high = +55.0°C)
                       (crit low =  +0.0°C, crit = +85.0°C)

spd5118-i2c-1-51
Adapter: SMBus PIIX4 adapter port 0 at 0b00
temp1:        +36.2°C  (low  =  +0.0°C, high = +55.0°C)
                       (crit low =  +0.0°C, crit = +85.0°C)

amdgpu-pci-0300
Adapter: PCI adapter
vddgfx:      699.00 mV 
fan1:           0 RPM  (min =    0 RPM, max = 3200 RPM)
edge:         +27.0°C  (crit = +100.0°C, hyst = -273.1°C)
                       (emerg = +105.0°C)
junction:     +33.0°C  (crit = +110.0°C, hyst = -273.1°C)
                       (emerg = +115.0°C)
mem:          +42.0°C  (crit = +108.0°C, hyst = -273.1°C)
                       (emerg = +113.0°C)
PPT:          43.00 W  (cap = 339.00 W)

nvme-pci-0f00
Adapter: PCI adapter
Composite:    +37.9°C  (low  = -20.1°C, high = +78.8°C)
                       (crit = +83.8°C)
Sensor 1:     +37.9°C  (low  = -273.1°C, high = +65261.8°C)
```

pvtri96, 2025-04-25:
Any chance for an update on this issue, I check out and build the branch 16-fans. However, after properly integrate the module, I only have access to the RPM values, but cannot control the RPM. I could give a hand for the integration/testing step, perhaps made some code changes too if any hint is given.

demomanca, 2025-04-25:
> Any chance for an update on this issue? I check out and build the branch 16-fans. However, after properly integrate the module, I only have access to the RPM values, but cannot control the RPM. I [...]

> Yeah, we haven't worked out the PWM control signal points for those fans yet. MSI and Nuvoton won't tell us where the register is. :(

Still no change since this comment. We haven't been able to find the control register for the system fans.

sgoth, 2025-04-25:
The original "upstream" did:

https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/1644

Signulled, 2026-01-14:
I'm not sure if my issue is related to this or I need to create a new issue, but I'm having similar issues.

Kernel Version: 6.18.5-2-cachyos (64-bit)
Motherboard: MSI PRO X870E-P WIFI (MS-7E70) Running latest bios version 7E70v2A33

Currently, only my CPU Fan plugged into the cpu fan header is able to be controlled. I have my case fans plugged into my sys fan 3 header via a hub (HAVN BF360 Flow). 

I'm testing through CoolerControl. The case fan is able to be manually set as well as operate off a graph curve. I've tested manually setting the system fan 3 manually, curve, and mix, and it is stuck at 60%. I've gone into bios and manually turned off the smart fan feature, set it to 0%, 100%, 15%, auto, etc.

Here is an output of sensors.

```
r8169_0_e00:00-mdio-0
Adapter: MDIO adapter
temp1:        +33.5°C  (high = +120.0°C)

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +42.9°C
Tccd1:        +31.4°C

nvme-pci-0200
Adapter: PCI adapter
Composite:    +28.9°C  (low  = -273.1°C, high = +81.8°C)
                       (crit = +84.8°C)
Sensor 1:     +28.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +36.9°C  (low  = -273.1°C, high = +65261.8°C)

nct6687-isa-0a20
Adapter: ISA adapter
+12V:           12.05 V  (min = +12.02 V, max = +12.05 V)
+5V:             5.04 V  (min =  +5.03 V, max =  +5.05 V)
+3.3V:           2.05 V  (min =  +0.00 V, max =  +3.36 V)
CPU Soc:         1.21 V  (min =  +1.21 V, max =  +1.21 V)
CPU Vcore:       1.15 V  (min =  +0.95 V, max =  +1.20 V)
CPU 1P8:         3.35 V  (min =  +3.34 V, max =  +3.36 V)
CPU VDDP:        3.35 V  (min =  +3.34 V, max =  +3.36 V)
DRAM:            1.36 V  (min =  +1.36 V, max =  +1.36 V)
Chipset:       950.00 mV (min =  +0.73 V, max =  +1.02 V)
CPU SA:        930.00 mV (min =  +0.71 V, max =  +1.01 V)
Voltage #2:      1.51 V  (min =  +1.51 V, max =  +1.51 V)
AVCC3:           3.28 V  (min =  +3.27 V, max =  +3.28 V)
AVSB:            3.09 V  (min =  +0.00 V, max =  +3.36 V)
VBat:            1.83 V  (min =  +0.00 V, max =  +2.05 V)
CPU Fan:        749 RPM  (min =    0 RPM, max = 2218 RPM)
Pump Fan:         0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #1:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #2:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #3:  917 RPM  (min =    0 RPM, max = 1034 RPM)
System Fan #4:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #5:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #6:    0 RPM  (min =    0 RPM, max =    0 RPM)
CPU:            +42.0°C  (low  = +41.0°C, high = +53.0°C)
System:         +29.0°C  (low  = +28.0°C, high = +39.0°C)
VRM MOS:        +27.5°C  (low  = +24.0°C, high = +36.0°C)
PCH:            +30.0°C  (low  = +25.5°C, high = +42.5°C)
CPU Socket:     +29.0°C  (low  = +25.0°C, high = +41.0°C)
PCIe x1:         +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
M2_1:            +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
pwm1:               30%  MANUAL CONTROL
pwm2:               77%
pwm3:               77%  MANUAL CONTROL
pwm4:               77%  MANUAL CONTROL
pwm5:               77%  MANUAL CONTROL
pwm6:               77%  MANUAL CONTROL
pwm7:               77%  MANUAL CONTROL
pwm8:               77%  MANUAL CONTROL

spd5118-i2c-13-53
Adapter: SMBus PIIX4 adapter port 0 at 0b00
temp1:        +23.5°C  (low  =  +0.0°C, high = +55.0°C)
                       (crit low =  +0.0°C, crit = +85.0°C)

amdgpu-pci-7a00
Adapter: PCI adapter
vddgfx:        1.15 V
vddnb:         1.20 V
edge:         +37.0°C
PPT:          11.00 mW
sclk:         600 MHz

nvme-pci-0500
Adapter: PCI adapter
Composite:    +25.9°C  (low  =  -5.2°C, high = +82.8°C)
                       (crit = +84.8°C)
Sensor 1:     +25.9°C  (low  = -273.1°C, high = +65261.8°C)
```

And here is grep dmesg | grep nct

```
[    2.892382] nvme nvme1: pci function 0000:05:00.0
[    2.892382] nvme nvme0: pci function 0000:02:00.0
[    4.265227] amdgpu: ATPX version 1, functions 0x00000000
[    6.268970] nct6687: module verification failed: signature and/or required key missing - tainting kernel
[    6.270651] nct6687: Detected MSI board requiring msi_alt1 fan configuration
[    7.304452] nct6687: Found NCT6687D or compatible chip at 0x004e:0x0a20
[    7.304488] nct6687 nct6687.2592: Detected MSI board; using alternative fan configuration (msi_alt1)
[    7.304489] nct6687 nct6687.2592: MSI fan brute force mode: disabled
[    7.310130] nct6687 nct6687.2592: NCT6687D EC firmware version 0.0 build 04/18/25
[   63.635668] nct6687: The device rejected new configuration as invalid
[  122.942796] nct6687: The device rejected new configuration as invalid
[  488.415361] nct6687: The device rejected new configuration as invalid
[  490.415413] nct6687: The device rejected new configuration as invalid
[  499.419563] nct6687: The device rejected new configuration as invalid
[  499.430570] nct6687: The device rejected new configuration as invalid
[  501.419617] nct6687: The device rejected new configuration as invalid
[  501.431616] nct6687: The device rejected new configuration as invalid
[  532.416827] nct6687: The device rejected new configuration as invalid
```

Here you can see I set it to 40%, but it's still running at 60%.

[image]

I apologize as I'm fairly new to linux. I'll try to provide as much information as I can. Thank you for your time.

PierrickViot-Launier, 2026-09-30:
I have an issue with the MSI MAG X870 TOMAHAWK WIFI: The EZ Conn Fan is not exposed

Hardware:
```
Motherboard: MSI MAG X870 TOMAHAWK WIFI (MS-7E51)
EC: NCT6687D
BIOS detects fans through EZ Conn (JAF_2)
```
Driver detection:
```
nct6687: Detected MSI board requiring msi_alt1 fan configuration
nct6687 nct6687.2592: Detected MSI board; using alternative fan configuration (msi_alt1)
nct6687 nct6687.2592: MSI fan brute force mode: disabled
nct6687 nct6687.2592: active fan config=msi_alt1, SYS_FAN reg_rpm=0x015E/0x015C/0x015A
nct6687 nct6687.2592: enabled channels: fan/pwm mask=0xff, temp mask=0x7f
nct6687 nct6687.2592: NCT6687D EC firmware version 0.0 build 08/22/24
```
Problem:
The motherboard BIOS exposes my case fans as EZ Conn Fan and SYS_FAN1-6 are unused

[image]

`sensors` output:
```
ath12k_hwmon-pci-0700
Adapter: PCI adapter
temp1:        +43.0°C  

spd5118-i2c-5-53
Adapter: SMBus PIIX4 adapter port 0 at 0b00
temp1:        +29.5°C  (low  =  +0.0°C, high = +55.0°C)
                       (crit low =  +0.0°C, crit = +85.0°C)

hidpp_battery_0-hid-3-a
Adapter: HID adapter
in0:           4.04 V  

r8169_0_800:00-mdio-0
Adapter: MDIO adapter
temp1:        +44.0°C  (high = +110.0°C)

prom21_xhci-pci-0b00
Adapter: PCI adapter
temp1:        +47.4°C  

nvme-pci-0a00
Adapter: PCI adapter
Composite:    +32.9°C  (low  = -273.1°C, high = +84.8°C)
                       (crit = +84.8°C)
Sensor 1:     +32.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +32.9°C  (low  = -273.1°C, high = +65261.8°C)

nct6687-isa-0a20
Adapter: ISA adapter
+12V:           12.02 V  (min = +11.98 V, max = +12.02 V)
+5V:             5.06 V  (min =  +5.06 V, max =  +5.07 V)
+3.3V:           3.35 V  (min =  +0.00 V, max =  +3.35 V)
CPU Soc:         1.22 V  (min =  +1.21 V, max =  +1.22 V)
CPU Vcore:     966.00 mV (min =  +0.86 V, max =  +1.27 V)
CPU 1P8:         3.35 V  (min =  +3.34 V, max =  +3.35 V)
CPU VDDP:        3.35 V  (min =  +3.34 V, max =  +3.35 V)
DRAM:            1.40 V  (min =  +1.40 V, max =  +1.41 V)
Chipset:       294.00 mV (min =  +0.28 V, max =  +0.33 V)
CPU SA:        818.00 mV (min =  +0.81 V, max =  +0.83 V)
Voltage #2:      1.51 V  (min =  +1.51 V, max =  +1.51 V)
AVCC3:           3.31 V  (min =  +3.30 V, max =  +3.31 V)
AVSB:            3.34 V  (min =  +0.00 V, max =  +3.35 V)
VBat:            2.04 V  (min =  +0.00 V, max =  +2.04 V)
CPU Fan:        562 RPM  (min =  561 RPM, max =  768 RPM)
Pump Fan:         0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #1:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #2:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #3:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #4:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #5:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #6:    0 RPM  (min =    0 RPM, max =    0 RPM)
CPU:            +44.0°C  (low  = +42.0°C, high = +53.0°C)
System:         +37.5°C  (low  = +37.0°C, high = +38.5°C)
VRM MOS:        +35.5°C  (low  = +35.5°C, high = +37.0°C)
PCH:            +36.0°C  (low  = +35.5°C, high = +36.5°C)
CPU Socket:      +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
PCIe x1:         +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
M2_1:            +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
pwm1:               44%
pwm2:               92%
pwm3:               44%
pwm4:               44%
pwm5:               44%
pwm6:               44%
pwm7:               44%
pwm8:               44%

amdgpu-pci-7100
Adapter: PCI adapter
vddgfx:      960.00 mV 
vddnb:         1.20 V  
edge:         +38.0°C  
PPT:          26.00 mW 
sclk:         600 MHz 

spd5118-i2c-5-51
Adapter: SMBus PIIX4 adapter port 0 at 0b00
temp1:        +30.5°C  (low  =  +0.0°C, high = +55.0°C)
                       (crit low =  +0.0°C, crit = +85.0°C)

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +44.4°C  
Tccd1:        +32.5°C  

nvme-pci-0200
Adapter: PCI adapter
Composite:    +36.9°C  (low  = -273.1°C, high = +84.8°C)
                       (crit = +84.8°C)
Sensor 1:     +36.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +29.9°C  (low  = -273.1°C, high = +65261.8°C)

```

The fans are functional and controlled correctly in Windows using FanControl + LibreHardwareMonitor.
Would it be possible to expose the MSI EZ Conn fan channel in the Linux driver?
The BIOS appears to expose this as a separate fan channel from SYS_FAN headers.
I can provide additional testing if needed.