Fred78290/nct6687d #108: Kernel interface errors?

recallmenot, 2024-10-30:
Hi,

I'm using https://aur.archlinux.org/nct6687d-dkms-git.git with the [config](https://github.com/Fred78290/nct6687d/blob/main/sensors.d/Z690_PRO_A.conf) I provided.
I'm getting the following output from sensors:

```
12V:           12.07 V  (min = +12.07 V, max = +12.07 V)
+5V:             4.97 V  (min =  +4.96 V, max =  +4.97 V)
Vcore:         828.00 mV (min =  +0.83 V, max =  +1.01 V)
VIN3:          830.00 mV (min =  +0.83 V, max =  +0.83 V)
DRAM:            1.42 V  (min =  +1.42 V, max =  +1.42 V)
CPU SA:        820.00 mV (min =  +0.82 V, max =  +0.82 V)
CPU AUX:         1.81 V  (min =  +1.81 V, max =  +1.81 V)
VIN7:            1.53 V  (min =  +1.53 V, max =  +1.53 V)
+3.3V:           3.31 V  (min =  +3.31 V, max =  +3.32 V)
CPU Fan:        455 RPM  (min =  455 RPM)
Pump Fan:       987 RPM  (min =  987 RPM)
System Fan #1:    0 RPM  (min =    0 RPM)
System Fan #2:  692 RPM  (min =  666 RPM)
System Fan #3:  600 RPM  (min =  600 RPM)
System Fan #4:    0 RPM  (min =    0 RPM)
System Fan #5:    0 RPM  (min =    0 RPM)
System Fan #6:    0 RPM  (min =    0 RPM)
ERROR: Can't get value of subfeature fan9_input: Kernel interface error
ERROR: Can't get value of subfeature fan9_min: Kernel interface error
fan9:               N/A  (min =    0 RPM)
ERROR: Can't get value of subfeature fan10_input: Kernel interface error
ERROR: Can't get value of subfeature fan10_min: Kernel interface error
fan10:              N/A  (min =    0 RPM)
ERROR: Can't get value of subfeature temp1_max_hyst: Kernel interface error
ERROR: Can't get value of subfeature temp1_crit: Kernel interface error
CPU:            +29.0°C  (low  = +29.0°C)
                         (high = +31.0°C, hyst =  +0.0°C)
ERROR: Can't get value of subfeature temp1_type: Kernel interface error
                         (crit =  +0.0°C)  sensor = disabled
ERROR: Can't get value of subfeature temp2_max_hyst: Kernel interface error
ERROR: Can't get value of subfeature temp2_crit: Kernel interface error
System:         +39.0°C  (low  = +38.0°C)
                         (high = +39.0°C, hyst =  +0.0°C)
ERROR: Can't get value of subfeature temp2_type: Kernel interface error
                         (crit =  +0.0°C)  sensor = disabled
ERROR: Can't get value of subfeature temp3_max_hyst: Kernel interface error
ERROR: Can't get value of subfeature temp3_crit: Kernel interface error
VRM MOS:        +32.5°C  (low  = +32.0°C)
                         (high = +32.5°C, hyst =  +0.0°C)
ERROR: Can't get value of subfeature temp3_type: Kernel interface error
                         (crit =  +0.0°C)  sensor = disabled
ERROR: Can't get value of subfeature temp4_max_hyst: Kernel interface error
ERROR: Can't get value of subfeature temp4_crit: Kernel interface error
PCH:            +61.0°C  (low  = +61.0°C)
                         (high = +61.0°C, hyst =  +0.0°C)
                         (crit =  +0.0°C)
ERROR: Can't get value of subfeature temp5_max_hyst: Kernel interface error
ERROR: Can't get value of subfeature temp5_crit: Kernel interface error
CPU Socket:     +32.0°C  (low  = +32.0°C)
                         (high = +32.0°C, hyst =  +0.0°C)
ERROR: Can't get value of subfeature temp5_type: Kernel interface error
                         (crit =  +0.0°C)  sensor = disabled
ERROR: Can't get value of subfeature temp6_max_hyst: Kernel interface error
ERROR: Can't get value of subfeature temp6_crit: Kernel interface error
PCIe x1:        +36.0°C  (low  = +36.0°C)
                         (high = +36.0°C, hyst =  +0.0°C)
ERROR: Can't get value of subfeature temp6_type: Kernel interface error
                         (crit =  +0.0°C)  sensor = disabled
ERROR: Can't get value of subfeature temp7_max_hyst: Kernel interface error
ERROR: Can't get value of subfeature temp7_crit: Kernel interface error
M2_1:           +31.5°C  (low  = +31.0°C)
                         (high = +31.5°C, hyst =  +0.0°C)
ERROR: Can't get value of subfeature temp7_type: Kernel interface error
                         (crit =  +0.0°C)  sensor = disabled
```
Do you have any idea what these Kernel interface errors are about and how we could get rid of them?

Anyways, thanks for making this awesome kernel module!

recallmenot, 2024-10-31:
Strange; after a reboot and a reinstall of the kernel module the errors have vanished.

closed 2024-10-31 by recallmenot

vladns, 2025-01-05:
I get the same error.