Fred78290/nct6687d #110: Works on ASRock B650E Taichi Lite

zmascetta, 2024-11-18:
Built with Fedora 41. Works great in CoolerControl. 

Note, this MOBO technically has an NCT6686 chip in it.

TsunamiMommy, 2025-04-28:
Did you force load it with any parameters? As I have the same motherboard and it reports a chip id of 0xd441. Does yours also report that as well?

zmascetta, 2025-04-28:
Yes, that is what is reported initially. [This thread](https://github.com/lm-sensors/lm-sensors/issues/499) on lm-sensors was able to identify that chip, and was what eventually led me to this driver. As per that thread, I believe if you force load with `modprobe nct6683 force=on` it will show up identified, but it has been a while and I'm a bit hazy.

Have you built this driver/added it and your chip is still not being identified?

TsunamiMommy, 2025-04-28:
I was just seeing what it took to identify the chip properly and to make sure I wasn't missing anything as far as parameters are concerned. I also was hoping to verify that using the force parameter worked for that chipset before enabling it.