Fred78290/nct6687d #111: linux 6.12 appears to have broken fan control

kode54, 2024-11-19:
Testing with an MSI X670E Gaming Plus Wifi, and booting with a 6.12 kernel, either Arch's or CachyOS's, the sensors work, but fan control has ceased functioning. Fans are stuck at the BIOS configured defaults.

Fred78290, 2024-11-19:
It's worked with a 6.8 kernel?

HBunnick, 2024-12-07:
Hi Fred78290

your module worked for me in kernel 6.12.1-arch1-1
and now it seems to be broken in kernel 6.12.3-arch1-1

i did not investigate if further, but the command
`make dkms/install`
does not run successfully.

Hope it's helpful intel for you.

My Mainboard is MSI MPG B550 GAMING PLUS (MS-7C90)

Fred78290, 2024-12-15:
> Hi Fred78290
>
> your module worked for me in kernel 6.12.1-arch1-1 and now it seems to be broken in kernel 6.12.3-arch1-1
>
> i did not investigate if further, but the command `make dkms/install` does [...]

Could you log the build error?