Fred78290/nct6687d #112: Not working fully on ASRock x870e Taichi (linux 6.12)

Juts2, 2024-11-20:
Unfortunately it seems the fan controls and RPM detection are not working for me. Not sure what I can try next.

Initially with sensors-detect I saw unknown chip with ID 0xd441

Now after installing, sensors detect shows this:

Found unknown chip with ID 0xd802
    (logical device B has address 0x290, could be sensors)

The sensors command does pick up a lot more than prior, but unfortunately all of the fan info is incorrect / not working

System fan 1 and system fan 2 should be reading RPM, same for CPU FAN.
Modifying the RPM manually with coolercontrol on these sensors didnt make any change. 

nct6686-isa-0a20
Adapter: ISA adapter
+12V:           12.67 V  (min = +12.67 V, max = +12.67 V)
+5V:             4.96 V  (min =  +4.96 V, max =  +4.96 V)
+3.3V:           3.42 V  (min =  +3.42 V, max =  +3.42 V)
CPU Soc:       992.00 mV (min =  +0.99 V, max =  +0.99 V)
CPU Vcore:       1.81 V  (min =  +1.81 V, max =  +1.81 V)
CPU 1P8:         3.42 V  (min =  +3.42 V, max =  +3.42 V)
CPU VDDP:        3.42 V  (min =  +3.42 V, max =  +3.42 V)
DRAM:            2.82 V  (min =  +2.78 V, max =  +2.82 V)
Chipset:       752.00 mV (min =  +0.75 V, max =  +0.75 V)
CPU SA:          1.68 V  (min =  +1.66 V, max =  +1.68 V)
Voltage #2:      1.12 V  (min =  +1.12 V, max =  +1.12 V)
AVCC3:           3.30 V  (min =  +3.30 V, max =  +3.30 V)
AVSB:            3.42 V  (min =  +3.42 V, max =  +3.42 V)
VBat:            1.81 V  (min =  +1.81 V, max =  +1.81 V)
CPU Fan:          0 RPM  (min =    0 RPM, max =    0 RPM)
Pump Fan:         0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #1:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #2:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #3:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #4:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #5:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #6:    0 RPM  (min =    0 RPM, max =    0 RPM)
CPU:            +46.0°C  (low  = +45.0°C, high = +53.0°C)
System:         +36.0°C  (low  = +36.0°C, high = +36.5°C)
VRM MOS:        +33.0°C  (low  = +33.0°C, high = +34.5°C)
PCH:            +47.0°C  (low  = +47.0°C, high = +47.0°C)
CPU Socket:     +66.0°C  (low  = +66.0°C, high = +66.0°C)
PCIe x1:        +62.0°C  (low  = +62.0°C, high = +62.0°C)
M2_1:           +62.0°C  (low  = +62.0°C, high = +62.0°C)

Fred78290, 2024-11-20:
your nct6686 is probably a variant...

try with native nct6683 to see if rpm aren't zero

Juts2, 2024-11-20:
> your nct6686 is probably a variant...
> 
> try with native nct6683 to see if rpm aren't zero

Removed the NCT6686 dkms module from this repository and then setup the module load for nct6683
❯ sudo sh -c 'echo "nct6683" >> /etc/modules-load.d/nct6683.conf'

After a reboot I do get a different sensor readout, but mostly the same issue. Thanks a lot for the help

❯ sensors
spd5118-i2c-7-51
Adapter: SMBus PIIX4 adapter port 0 at 0b00
temp1:        +33.2°C  (low  =  +0.0°C, high = +55.0°C)
                       (crit low =  +0.0°C, crit = +85.0°C)

nct6686-isa-0a20
Adapter: ISA adapter
VIN16:         752.00 mV (min =  +0.00 V, max =  +0.00 V)
VIN0:            1.06 V  (min =  +0.00 V, max =  +0.00 V)
VIN1:          992.00 mV (min =  +0.00 V, max =  +0.00 V)
VIN2:          992.00 mV (min =  +0.00 V, max =  +0.00 V)
VIN3:            1.68 V  (min =  +0.00 V, max =  +0.00 V)
VIN5:            1.12 V  (min =  +0.00 V, max =  +0.00 V)
VIN6:            1.41 V  (min =  +0.00 V, max =  +0.00 V)
VIN7:            1.81 V  (min =  +0.00 V, max =  +0.00 V)
VIN0:            1.06 V  (min =  +0.00 V, max =  +0.00 V)
VIN1:          992.00 mV (min =  +0.00 V, max =  +0.00 V)
VIN2:          992.00 mV (min =  +0.00 V, max =  +0.00 V)
VIN6:            1.41 V  (min =  +0.00 V, max =  +0.00 V)
VIN7:            1.81 V  (min =  +0.00 V, max =  +0.00 V)
VIN16:         752.00 mV (min =  +0.00 V, max =  +0.00 V)
VIN3:            1.68 V  (min =  +0.00 V, max =  +0.00 V)
VIN5:            1.12 V  (min =  +0.00 V, max =  +0.00 V)
VCC:             3.30 V  (min =  +0.00 V, max =  +0.00 V)
VSB:             3.42 V  (min =  +0.00 V, max =  +0.00 V)
AVSB:            3.42 V  (min =  +0.00 V, max =  +0.00 V)
VSB:             3.42 V  (min =  +0.00 V, max =  +0.00 V)
AVSB:            3.42 V  (min =  +0.00 V, max =  +0.00 V)
VTT:             1.81 V  (min =  +0.00 V, max =  +0.00 V)
VBAT:            3.26 V  (min =  +0.00 V, max =  +0.00 V)
VREF:           48.00 mV (min =  +0.00 V, max =  +0.00 V)
fan1:             0 RPM  (min =    0 RPM)
fan2:             0 RPM  (min =    0 RPM)
fan3:             0 RPM  (min =    0 RPM)
fan4:             0 RPM  (min =    0 RPM)
fan5:             0 RPM  (min =    0 RPM)
fan7:             0 RPM  (min =    0 RPM)
SMBus 0:        +46.0°C  (low  =  +0.0°C)
                         (high =  +0.0°C, hyst =  +0.0°C)
                         (crit =  +0.0°C)
Thermistor 14:  +36.0°C  (low  =  +0.0°C)
                         (high =  +0.0°C, hyst =  +0.0°C)
                         (crit =  +0.0°C)  sensor = thermistor
Thermistor 15:  +33.5°C  (low  =  +0.0°C)
                         (high =  +0.0°C, hyst =  +0.0°C)
                         (crit =  +0.0°C)  sensor = thermistor
SMBus 1:        -62.0°C  (low  =  +0.0°C)
                         (high =  +0.0°C, hyst =  +0.0°C)
                         (crit =  +0.0°C)
SMBus 2:        -62.0°C  (low  =  +0.0°C)
                         (high =  +0.0°C, hyst =  +0.0°C)
                         (crit =  +0.0°C)
SMBus 3:        -62.0°C  (low  =  +0.0°C)
                         (high =  +0.0°C, hyst =  +0.0°C)
                         (crit =  +0.0°C)
intrusion0:    OK
beep_enable:   disabled

Juts2, 2024-11-20:
Ok, so trying out different modules nct6775 worked for fan sensors & control on this board. I dont know if its just similar or if thats the module I have onboard, but I am happy it is working.