Fred78290/nct6687d #115: PRO H610M-E DDR4 (MS-7D48)

axet, 2025-02-10:
Hello!

Thanks for supporting MSI products! Your driver loaded but produce incorrect output. 

[image]

This board chip is: NUVOTON NCT6687D-M

It seems module reports incorrect voltage, not fan speed.

```
nct6687-isa-0a20
Adapter: ISA adapter
VIN0:             1.02 V  (min =  +0.00 V, max =  +0.00 V)
VIN1:           992.00 mV (min =  +0.00 V, max =  +0.00 V)
VIN2:           704.00 mV (min =  +0.00 V, max =  +0.00 V)
VIN3:           416.00 mV (min =  +0.00 V, max =  +0.00 V)
VIN4:           592.00 mV (min =  +0.00 V, max =  +0.00 V)
VIN5:           928.00 mV (min =  +0.00 V, max =  +0.00 V)
VIN6:           896.00 mV (min =  +0.00 V, max =  +0.00 V)
VIN7:             1.54 V  (min =  +0.00 V, max =  +0.00 V)
VCC:              3.34 V  (min =  +0.00 V, max =  +0.00 V)
fan1:              0 RPM  (min =    0 RPM)
fan2:              0 RPM  (min =    0 RPM)
fan3:              0 RPM  (min =    0 RPM)
fan4:              0 RPM  (min =    0 RPM)
fan5:              0 RPM  (min =    0 RPM)
fan6:              0 RPM  (min =    0 RPM)
fan7:              0 RPM  (min =    0 RPM)
fan8:              0 RPM  (min =    0 RPM)
fan9:              0 RPM  (min =    0 RPM)
fan10:             0 RPM  (min =    0 RPM)
PECI 0.0:        +48.5°C  (low  =  +0.0°C)
                          (high =  +0.0°C, hyst =  +0.0°C)
                          (crit =  +0.0°C)  sensor = Intel PECI
Diode 0 (curr):  +42.0°C  (low  =  +0.0°C)
                          (high =  +0.0°C, hyst =  +0.0°C)
                          (crit =  +0.0°C)  sensor = thermal diode
Thermistor 15:   +40.0°C  (low  =  +0.0°C)
                          (high = +112.0°C, hyst = +12.0°C)
                          (crit =  +0.0°C)  sensor = thermistor
PCH CHIP:        +71.0°C  (low  =  +0.0°C)
                          (high =  +0.0°C, hyst =  +0.0°C)
                          (crit =  +0.0°C)
Thermistor 16:   +45.5°C  (low  =  +0.0°C)
                          (high =  +0.0°C, hyst =  +0.0°C)
                          (crit =  +0.0°C)  sensor = thermistor
Thermistor 0:    +30.0°C  (low  =  +0.0°C)
                          (high =  +0.0°C, hyst =  +0.0°C)
                          (crit =  +0.0°C)  sensor = thermistor
Thermistor 1:    +31.5°C  (low  =  +0.0°C)
                          (high =  +0.0°C, hyst =  +0.0°C)
                          (crit =  +0.0°C)  sensor = thermistor
intrusion0:     OK
beep_enable:    disabled
```

Fred78290, 2025-02-10:
Hello I think that you have the nct6683 module running not mine because my module doesn't show 10 fans but 8 :) and i don't report beep_enable and intrusion.

you must blacklisted nct6683 and load nct6687

axet, 2025-02-10:
Right. I'm so bad. It works! 😅

```
nct6687-isa-0a20
Adapter: ISA adapter
+12V:           12.22 V  (min = +12.22 V, max = +12.36 V)
+5V:             5.07 V  (min =  +5.03 V, max =  +5.07 V)
+3.3V:           3.34 V  (min =  +0.00 V, max =  +3.34 V)
CPU Soc:         1.09 V  (min =  +0.72 V, max =  +1.09 V)
CPU Vcore:     600.00 mV (min =  +0.60 V, max =  +0.60 V)
CPU 1P8:         0.00 V  (min =  +0.00 V, max =  +0.00 V)
CPU VDDP:        0.00 V  (min =  +0.00 V, max =  +0.00 V)
DRAM:            1.24 V  (min =  +0.78 V, max =  +1.26 V)
Chipset:       936.00 mV (min =  +0.93 V, max =  +0.94 V)
CPU SA:        904.00 mV (min =  +0.90 V, max =  +0.91 V)
Voltage #2:      1.54 V  (min =  +1.53 V, max =  +1.54 V)
AVCC3:           3.34 V  (min =  +3.33 V, max =  +3.36 V)
AVSB:            3.34 V  (min =  +0.00 V, max =  +3.34 V)
VBat:            1.06 V  (min =  +0.00 V, max =  +1.06 V)
CPU Fan:       1996 RPM  (min =    0 RPM, max = 22222 RPM)
Pump Fan:         0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #1:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #2:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #3:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #4:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #5:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #6:    0 RPM  (min =    0 RPM, max =    0 RPM)
CPU:            +74.0°C  (low  = +47.0°C, high = +76.0°C)
System:         +42.5°C  (low  = +41.0°C, high = +42.5°C)
VRM MOS:        +46.5°C  (low  = +38.0°C, high = +46.5°C)
PCH:            +70.0°C  (low  = +68.0°C, high = +70.0°C)
CPU Socket:     +48.0°C  (low  = +44.0°C, high = +48.0°C)
PCIe x1:        +29.5°C  (low  = +29.5°C, high = +30.0°C)
M2_1:           +31.5°C  (low  = +31.0°C, high = +31.5°C)
```