Fred78290/nct6687d #119: Fancontrol not working on Linux 6.11.11

wxxsfxyzm, 2025-04-01:
I'm using Proxmox VE with latest edge kernel 6.11.11, along with a board MSI MAG b550m MORTAR, the sensor is nct6687.
I'm using default kernel module 'nct6683' shipped with the kernel and it worked well.
Until recently, I have updated kernel to 6.11.11-2-pve, and pwm control stopped working which said 'permission denied' when I want to change value of pwm node. So I searched and found this repository, after installed your module as `dkms`:
```bash
root@pve:~# dkms status
nct6687d/1, 6.11.11-2-pve, x86_64: installed
```
`sensors` command started working again
```bash
nct6687-isa-0a20
Adapter: ISA adapter
+12V:           12.07 V  (min = +12.07 V, max = +12.10 V)
+5V:             5.12 V  (min =  +5.11 V, max =  +5.13 V)
+3.3V:           3.34 V  (min =  +3.34 V, max =  +3.35 V)
CPU Soc:         1.11 V  (min =  +1.10 V, max =  +1.11 V)
CPU Vcore:       1.32 V  (min =  +0.88 V, max =  +1.34 V)
CPU 1P8:         1.85 V  (min =  +1.85 V, max =  +1.85 V)
CPU VDDP:        0.00 V  (min =  +0.00 V, max =  +0.00 V)
DRAM:            1.37 V  (min =  +1.37 V, max =  +1.37 V)
Chipset:         1.04 V  (min =  +1.01 V, max =  +1.04 V)
CPU SA:        906.00 mV (min =  +0.88 V, max =  +0.91 V)
Voltage #2:      1.54 V  (min =  +1.54 V, max =  +1.55 V)
AVCC3:           3.38 V  (min =  +3.38 V, max =  +3.39 V)
AVSB:            3.34 V  (min =  +3.34 V, max =  +3.34 V)
VBat:            1.85 V  (min =  +1.85 V, max =  +1.85 V)
CPU Fan:       1664 RPM  (min = 1558 RPM, max = 1889 RPM)
Pump Fan:         0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #1: 1177 RPM  (min =    0 RPM, max = 1224 RPM)
System Fan #2:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #3:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #4:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #5:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #6:    0 RPM  (min =    0 RPM, max =    0 RPM)
CPU:            +38.0°C  (low  = +34.0°C, high = +51.0°C)
System:         +30.5°C  (low  = +30.0°C, high = +32.0°C)
VRM MOS:        +25.0°C  (low  = +24.5°C, high = +26.5°C)
PCH:            +31.5°C  (low  = +31.0°C, high = +32.5°C)
CPU Socket:     +25.0°C  (low  = +24.5°C, high = +29.0°C)
PCIe x1:        +24.0°C  (low  = +24.0°C, high = +26.0°C)
M2_1:            +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
```
however `sensors-detect` and `pwmconfig` still not working.
when I run `sensors-detect` , it still tells me `Found unknown chip with ID 0xd592`:
```bash
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no):
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found unknown chip with ID 0xd592
```
when I run `pwmconfig`, the output will be very strage:
```bash
root@pve:~# pwmconfig
# pwmconfig version 3.6.0
This program will search your sensors for pulse width modulation (pwm)
controls, and test each one to see if it controls a fan on
your motherboard. Note that many motherboards do not have pwm
circuitry installed, even if your sensor chip supports pwm.

We will attempt to briefly stop each fan using the pwm controls.
The program will attempt to restore each fan to full speed
after testing. However, it is ** very important ** that you
physically verify that the fans have been to full speed
after the program has completed.

Found the following devices:
   hwmon0 is nvme
   hwmon1 is nct6687
   hwmon2 is nvme
   hwmon3 is k10temp
   hwmon5 is amdgpu

Found the following PWM controls:
   hwmon1/pwm1           current value: 98
hwmon1/pwm1 is currently setup for automatic speed control.
In general, automatic mode is preferred over manual mode, as
it is more efficient and it reacts faster. Are you sure that
you want to setup this output for manual control? (n) n
   hwmon1/pwm2           current value: 255
hwmon1/pwm2 is currently setup for automatic speed control.
In general, automatic mode is preferred over manual mode, as
it is more efficient and it reacts faster. Are you sure that
you want to setup this output for manual control? (n) n
   hwmon1/pwm3           current value: 153
hwmon1/pwm3 is currently setup for automatic speed control.
In general, automatic mode is preferred over manual mode, as
it is more efficient and it reacts faster. Are you sure that
you want to setup this output for manual control? (n) y
hwmon1/pwm3_enable stuck to 99
```
it does not test all fans but asked me directly, and when I entered `y` for my fan, it tells me `hwmon1/pwm3_enable stuck to 99`.
then the fan data will be lost from `sensors` output:
```bash
nct6687-isa-0a20
Adapter: ISA adapter
+12V:           12.10 V  (min = +12.07 V, max = +12.10 V)
+5V:             5.12 V  (min =  +5.11 V, max =  +5.13 V)
+3.3V:           3.34 V  (min =  +3.34 V, max =  +3.35 V)
CPU Soc:         1.11 V  (min =  +1.10 V, max =  +1.11 V)
CPU Vcore:     952.00 mV (min =  +0.88 V, max =  +1.34 V)
CPU 1P8:         1.85 V  (min =  +1.85 V, max =  +1.85 V)
CPU VDDP:        0.00 V  (min =  +0.00 V, max =  +0.00 V)
DRAM:            1.37 V  (min =  +1.37 V, max =  +1.37 V)
Chipset:         1.02 V  (min =  +1.01 V, max =  +1.04 V)
CPU SA:        892.00 mV (min =  +0.88 V, max =  +0.91 V)
Voltage #2:      1.54 V  (min =  +1.54 V, max =  +1.55 V)
AVCC3:           3.39 V  (min =  +3.38 V, max =  +3.39 V)
AVSB:            3.34 V  (min =  +3.34 V, max =  +3.34 V)
VBat:            1.85 V  (min =  +1.85 V, max =  +1.85 V)
CPU Fan:       1637 RPM  (min = 1558 RPM, max = 1889 RPM)
Pump Fan:         0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #1:    0 RPM  (min =    0 RPM, max = 1224 RPM)
System Fan #2:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #3:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #4:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #5:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #6:    0 RPM  (min =    0 RPM, max =    0 RPM)
CPU:            +38.0°C  (low  = +34.0°C, high = +51.0°C)
System:         +31.0°C  (low  = +30.0°C, high = +32.0°C)
VRM MOS:        +25.5°C  (low  = +24.5°C, high = +26.5°C)
PCH:            +32.0°C  (low  = +31.0°C, high = +32.5°C)
CPU Socket:     +26.5°C  (low  = +24.5°C, high = +29.0°C)
PCIe x1:        +25.0°C  (low  = +24.0°C, high = +26.0°C)
M2_1:            +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
```
**Notice System Fan 1 now become "0 RPM", the driver can't get it's actual data afterwards.**
at this time, only reloading nct6687 can help
```bash
modprobe -r nct6687
modprobe nct6687
```

I don't know whether this is caused by kernel itself or the driver side, so I am sending issues here for help!

Fred78290, 2025-04-01:
Did you activate secure boot ?

wxxsfxyzm, 2025-04-01:
No.
```bash
# dmesg | grep -i "secure boot"
[    0.000000] secureboot: Secure boot disabled
[    0.003739] secureboot: Secure boot disabled
```

wxxsfxyzm, 2025-04-01:
other information:
```bash
/sys/devices/platform/nct6687.2592/hwmon/hwmon1# grep -r . pwm*
pwm1:101
pwm1_enable:99
pwm2:255
pwm2_enable:99
pwm3:153
pwm3_enable:99
pwm4:153
pwm4_enable:99
pwm5:153
pwm5_enable:99
pwm6:255
pwm6_enable:99
pwm7:255
pwm7_enable:99
pwm8:128
pwm8_enable:99
```
only pwm1 and pwm3 connected to fans

xaizek, 2025-04-01:
`pwm*_enable` is for "fan speed control method" ([docs](https://www.kernel.org/doc/html/latest/hwmon/sysfs-interface.html#pwm)). `pwmconfig` seems to confuse it for `pwm*`.

wxxsfxyzm, 2025-04-02:
> `pwm*_enable` is for "fan speed control method" ([docs](https://www.kernel.org/doc/html/latest/hwmon/sysfs-interface.html#pwm)). `pwmconfig` seems to confuse it for `pwm*`.

you mean it is `lm-sensors` side problem? I noticed that it's repo hasn't been updated for years, how could I solve this?

xaizek, 2025-04-02:
> you mean it is `lm-sensors` side problem? I noticed that it's repo hasn't been updated for years, how could I solve this?

It looks like that to me.  There is at least one related issue https://github.com/lm-sensors/lm-sensors/issues/403 which suggests to edit config manually instead of using `pwmconfig`, that seems to entail creation of `/etc/fancontrol` after reading `man fancontrol`.

wxxsfxyzm, 2025-04-22:
after installing proxmox-kernel-6.14.0, fancontrol get back to work again.