Fred78290/nct6687d #123: Differences between nct6683 and nct6687 modules for nct6686d chip on Asrock B650 Steel Legend Wifi Motherboard

darkinaut, 2025-04-29:
Hi, just trying to provide information in order to improve support.

As stated by some other users, using kernel module nct6683 displays sensors information correctly but doesn't allow manually adjusting fan speeds, whereas this nct6687 module allows fan speeds adjustment but displays sensors info wrongly.

#modprobe nct6687
nct6687: Found NCT6686D or compatible chip at 0x002e:0x0a20
nct6687 nct6687.2592: NCT6686D EC firmware version 1.0 build 11/09/23

#pwmconfig output:
Found unknown chip with ID 0xd441

#sensors output when using kernel nct6683 module:
nct6686-isa-0a20
Adapter: ISA adapter
VIN0:               1.12 V  (min =  +0.00 V, max =  +0.00 V)
VIN1:               1.01 V  (min =  +0.00 V, max =  +0.00 V)
VIN2:               1.01 V  (min =  +0.00 V, max =  +0.00 V)
VIN3:               1.68 V  (min =  +0.00 V, max =  +0.00 V)
VIN5:               1.22 V  (min =  +0.00 V, max =  +0.00 V)
VIN6:               1.12 V  (min =  +0.00 V, max =  +0.00 V)
VIN7:               1.34 V  (min =  +0.00 V, max =  +0.00 V)
VCC:                3.36 V  (min =  +0.00 V, max =  +0.00 V)
VIN16:            432.00 mV (min =  +0.00 V, max =  +0.00 V)
fan1:              759 RPM  (min =    0 RPM)
fan2:              556 RPM  (min =    0 RPM)
fan3:              537 RPM  (min =    0 RPM)
fan4:              529 RPM  (min =    0 RPM)
fan5:              448 RPM  (min =    0 RPM)
fan6:              381 RPM  (min =    0 RPM)
Thermistor 14:     +40.5°C  (low  =  +0.0°C)
                            (high =  +0.0°C, hyst =  +0.0°C)
                            (crit =  +0.0°C)  sensor = thermistor
Thermistor 15:     +46.5°C  (low  =  +0.0°C)
                            (high =  +0.0°C, hyst =  +0.0°C)
                            (crit =  +0.0°C)  sensor = thermistor
AMD TSI Addr 98h:  +58.0°C  (low  =  +0.0°C)
                            (high =  +0.0°C, hyst =  +0.0°C)
                            (crit =  +0.0°C)  sensor = AMD AMDSI
intrusion0:       OK
beep_enable:      disabled

#sensors output when using kernel nct6683 module:
nct6686-isa-0a20
Adapter: ISA adapter
+12V:            0.00 V  (min =  +0.00 V, max =  +0.00 V)
+5V:             0.00 V  (min =  +0.00 V, max =  +0.00 V)
+3.3V:           3.42 V  (min =  +3.42 V, max =  +3.42 V)
CPU Soc:         0.00 V  (min =  +0.00 V, max =  +0.00 V)
CPU Vcore:       0.00 V  (min =  +0.00 V, max =  +0.00 V)
CPU 1P8:         0.00 V  (min =  +0.00 V, max =  +0.00 V)
CPU VDDP:        0.00 V  (min =  +0.00 V, max =  +0.00 V)
DRAM:            0.00 V  (min =  +0.00 V, max =  +0.00 V)
Chipset:         0.00 V  (min =  +0.00 V, max =  +0.00 V)
CPU SA:          0.00 V  (min =  +0.00 V, max =  +0.00 V)
Voltage #2:      0.00 V  (min =  +0.00 V, max =  +0.00 V)
AVCC3:           0.00 V  (min =  +0.00 V, max =  +0.00 V)
AVSB:            3.42 V  (min =  +3.42 V, max =  +3.42 V)
VBat:            1.86 V  (min =  +1.86 V, max =  +1.86 V)
CPU Fan:        655 RPM  (min =  647 RPM, max =  659 RPM)
Pump Fan:       558 RPM  (min =  555 RPM, max =  566 RPM)
System Fan #1:  532 RPM  (min =  532 RPM, max =  543 RPM)
System Fan #2:  526 RPM  (min =  526 RPM, max =  526 RPM)
System Fan #3:  457 RPM  (min =  455 RPM, max =  459 RPM)
System Fan #4:  381 RPM  (min =  377 RPM, max =  381 RPM)
System Fan #5:    0 RPM  (min =    0 RPM, max =    0 RPM)
System Fan #6:    0 RPM  (min =    0 RPM, max =    0 RPM)
CPU:             +0.0°C  (low  =  +0.0°C, high =  +0.0°C)
System:         +39.0°C  (low  = +39.0°C, high = +39.0°C)
VRM MOS:        +43.0°C  (low  = +43.0°C, high = +43.0°C)
PCH:            +53.0°C  (low  = +53.0°C, high = +54.0°C)
CPU Socket:     +69.0°C  (low  = +69.0°C, high = +69.0°C)
PCIe x1:        +63.0°C  (low  = +63.0°C, high = +63.0°C)
M2_1:           +63.0°C  (low  = +63.0°C, high = +63.0°C)

Personally I favor the nct6687 module since I really need to adjust fan speeds manually, but I have no idea how to create a custom sensors.conf based on what nct6683 module is displaying....glad if somebody could help improve this. Thank you

Fred78290, 2025-04-30:
It's the same as https://github.com/Fred78290/nct6687d?tab=readme-ov-file#voltage-manual-configuration

s25g5d4, 2025-12-14:
I had a fork to deal with the nct6686d on ASRock motherboards: https://github.com/s25g5d4/nct6686d
mine is A620I Lightning WiFi. AFAIK, the layout is almost the same as your B650 Steel Legend WiFi, so you can give it a try.
however, I'm still running on kernel version 6.8; the newer kernel has some breaking changes, and I have not yet had time to adapt it.