Fred78290/nct6687d #128: Not building on Arch Linux

xiyori, 2025-06-16:
After Jun 15 merges the module does not build anymore on Arch.

```
==> dkms install --no-depmod nct6687d-dkms-git/r119.7df2680 -k 6.15.2-arch1-1

Error! Build of nct6687.ko failed for: 6.15.2-arch1-1 (x86_64)
Make sure the name and location of the generated module are correct,
or consult /var/lib/dkms/nct6687d-dkms-git/r119.7df2680/build/make.log for more information.
==> WARNING: `dkms install --no-depmod nct6687d-dkms-git/r119.7df2680 -k 6.15.2-arch1-1' exited 7
```

make.log says

```
DKMS (dkms-3.2.1) make.log for nct6687d-dkms-git/r119.7df2680 for kernel 6.15.2-arch1-1 (x86_64)
Mon Jun 16 09:53:45 AM MSK 2025

Building module(s)
# command: make -j32 KERNELRELEASE=6.15.2-arch1-1 -C /usr/lib/modules/6.15.2-arch1-1/build M=/var/lib/dkms/nct6687d-dkms-git/r119.7df2680/build
make: Entering directory '/usr/lib/modules/6.15.2-arch1-1/build'
make[1]: Entering directory '/var/lib/dkms/nct6687d-dkms-git/r119.7df2680/build'
  MODPOST Module.symvers
make[1]: Leaving directory '/var/lib/dkms/nct6687d-dkms-git/r119.7df2680/build'
make: Leaving directory '/usr/lib/modules/6.15.2-arch1-1/build'

# exit code: 0
# elapsed time: 00:00:01
```

xaizek, 2025-06-16:
I had similar symptoms which meant that the build was happening in a wrong directory (this is how dkms reports "no nct6687.ko file found") and IIRC dropping `MAKE[0]` fixed that, so I would check that your `dkms.conf` was updated as in https://github.com/Fred78290/nct6687d/commit/ebbe59b205c971457139ff512a89a424ee80fe3f.

kode54, 2025-06-16:
Patch needed for AUR package, since it was still filtering and altering the distribution Makefile:

https://gist.github.com/kode54/e497014cbbfc45a1014b5e266f9a61b2

kode54, 2025-06-17:
> [@kode54](https://github.com/kode54) are you able to create a PR?

AUR doesn't support PRs, one of the maintainers must push the changes directly. Unless the repository is also duplicated on Github or something.

closed 2025-06-19 by Fred78290

kode54, 2025-06-19:
You didn't have to do that. I proposed the changes necessary to the AUR package to fix it. Unfortunately, one has to wait up to a month before a package may be orphaned. That's:

1) File a comment or email to the maintainer
2) Wait two weeks for a result
3) This failing, file an orphan request
4) Wait two more weeks for the maintainer to respond
5) That failing, package is orphaned.

PR #130 (2025-06-19): Bring back copying of Makefile as is
Instead just silence errors from `git` to not see confusing output.

Closes #128, closes #127.