Fred78290/nct6687d #136: Works on MSI PRO B850-P WiFi, with tweaks

ppq556, 2025-08-12:
I wanted to say "Thank you!" for developing and maintaining this module, it works on my MSI PRO B850-P WiFi.

NCT6687D is onboard:

nct6687: Found NCT6687D or compatible chip at 0x004e:0x0a20

The only tweak I had to do is to increase the number of fan channels to 16:

-#define NCT6687_NUM_REG_FAN 8
-#define NCT6687_NUM_REG_PWM 8
+#define NCT6687_NUM_REG_FAN 16
+#define NCT6687_NUM_REG_PWM 16

It turns out that my system fans 1 & 2 are in fact 15 and 16 using device indexes.

/etc/sensors.d/MSI_B850-P.conf:
chip "nct6687-*"
	label fan15 "System Fan 1"
	label fan16 "System Fan 2"
	ignore fan3
	ignore fan4

Thanks again!

kattkaries, 2025-08-13:
Hey!  B850 Tomahawk Max Wifi here. Thanks for posting your findings! Where and how did you tweak it to increase the number of fan channels?

kattkaries, 2025-08-15:
Hey again! No dice, sadly! I've changed the number of fan channels to 16, but still only the CPU fan is identified correctly. Strangely VRM MOS reports a fluctuating RPM, which of course isn't  affected by manually setting "fan speed" in CoolerControl.

Related to issue [107](https://github.com/Fred78290/nct6687d/issues/107).

ppq556, 2025-08-15:
@kattkaries RPMs in "VRM MOS" means the tweak is working, just the default label is incorrect.

Run "sensors -A -u" to see what is the true name of the series returning these RPMs. High chance it's in fact fan15 or fan16.
If so, do the simple fix in the sensors.conf:

$ cat /etc/sensors.d/MSI_B850-P.conf 
chip "nct6687-*"
	label fan15 "System Fan 1"
	label fan16 "System Fan 2"

kattkaries, 2025-08-16:
Yes, it was partly a labeling issue! Thanks for the help! <3 But to be clear, only the CPU fan is PWM-controllable for me. The others give kernel interface errors like so:

Top Out:
ERROR: Can't get value of subfeature fan14_input: Kernel interface error
ERROR: Can't get value of subfeature fan14_min: Kernel interface error
ERROR: Can't get value of subfeature fan14_pulses: Kernel interface error

Also, the values jump to ridiculous values intermittently (like every 3 seconds or so) of 65535 or the like, so it seems there are other issues as well. :P

Guess I'll have to rely on BIOS curves for the time being, until #107  is fixed.